Ultimaker/Cura #6672: Alternate direction of nozzle rotation.

ffaille, 2019-11-17:
I've noticed that with some nozzle (reprappro quickset nozzle is the best example) print defaults due to plastic accumulation (mainly when printing a little above nothing - starting with a angle near 30%) are always on the same side of the printed part.

Could you add an option to alternate the direction of rotation of the nozzle to each layer for wall printing ? This will prevent the accumulation of plastic and distribute the remaining printing defects over the entire part.

I have, of course, tried to better calibrate and adjust everything that could be calibrated. I have tested different qualities of plastics at different temperatures with different printing speeds. The problem observed is that the plastic remains "stuck" to the nozzle.

I have noticed these kinds of problems when the nozzles are not very sharp.

I think that this option has the potential to improve the rendering in any case.

rburema, 2019-11-19:
Have you tried putting the 'Z Seam Alignment' to 'Random'?

ffaille, 2019-11-19:
Yes, this is not a Z-seam problem related.

Ghostkeeper, 2019-12-04:
I'm not really sure how alternating the direction of printing walls would fix your overextrusion problem though.

jf3303, 2020-05-27:
I would also like to alternate the direction of rotation. I think it would help to make perfect circle with cheap printers. Because of small loose in the printer parts I see crack in thin cylinders between internal and external shell.

Ghostkeeper, 2020-05-29:
But if you switch directions of the wall lines you would see more cracks, due to hysteresis, right?

jf3303, 2020-05-29:
The defect would be on the opposite side from the previous layer each time. Structural integrity would be optimized at the cost of an inferior surface finish.

Ghostkeeper, 2020-06-03:
If the Seam settings still work the same way, the seam would still be placed in the same location as it is now, not alternating from the previous layer.

dnasir, 2020-12-10:
I think this would be useful when dealing with overhangs as well. I noticed that my overhangs tend to slightly curl up on one side, usually at the seams.

Then again, I probably just need better cooling.

rburema, 2023-05-17:
Already implemented as _"Alternate Wall Directions"_, under the _'Experimental'_ heading.

closed 2023-05-17 by rburema

Zomflax, 2023-05-18:
> 

Thank you, that is indeed a nice implementation! However, at first we were thinking of alternating the wall directions from one layer to another, but your option does more than that and alternate each printed lines, which is great! Then, would it be possible to have the lines of the same layer go in the same direction, but the next layer would have them in the opposite direction?

rburema, 2023-05-22:
@Zomflax You're welcome! Currently, as written, the behaviours are indeed combined. Is this a problem in some way?

Zomflax, 2023-05-22:
@rburema Actually, I noticed alternating the wall direction on the same layer results in a more "messy" surface, not really flat, which is pretty logic. So having the same wall direction on a layer would let the next layer to have a flat surface to stick on. But still having the walls going in the opposite direction for the next layer may help layer adhesion.
NB: Even without this option on, when doing cylinders, sometimes the walls alternate directions anyway on the same layer, so the "messy" surface can still appear...