Ultimaker/Cura #6673: Velleman Vertex K8600 printer definition wrong

Leleu256, 2019-11-17:
**Application version**
(The version of the application this issue occurs with.)

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

**Reproduction steps**
1. Something you did.
2. Something you did next.

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

Leleu256, 2019-11-17:
Checking the vertex_nano_k8600.def.json figure where is the issue. The file does not include "\n" special character.
Also, the nozzle is defined to 0.35mm but the stock model use a 0,4mm nozzle.

I've tried to correct the file but for a reason I ignore, even if I modify the file, cura ignore my modifications.

What I propose is as below : 

    "overrides": {
        "machine_start_gcode": {
            "default_value": "M104 T0 S{material_print_temperature} \nG28 ; Home extruder\nG90 ; Absolute positioning\nM82 ; Extruder in absolute mode\nG92 E0 ; Reset extruder position\nM109 T0 S{material_print_temperature} \nG1 E20 F100\nG92 E0 ; Reset extruder position\n"
        },
        "machine_end_gcode": {
            "default_value": "; Default end code\nG1 X0 Y0 Z130 ; Get extruder out of way. Uncomment to use!\nM107 ; Turn off fan\n; Disable all extruder\nG91 ; Relative positioning\nT0\nG1 E-1 ; Reduce filament pressure\nM104 T0 S0\nG90 ; Absolute positioning\nG92 E0 ; Reset extruder position\nM140 S0 ; Disable heated bed\nM84 ; Turn steppers off\n"
        },
        "infill_line_width": {
            "value": 0.4
        },
        "wall_thickness": {
            "value": 0.8
        },
}

Leleu256, 2019-11-17:
The default start gcode by the manufacturer is  : 

{IF_EXT0}M104 T0 S{TEMP0}
G28 ; Home extruder
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
G1 Z1 F{Z_TRAVEL_SPEED}
G92 E0 ; Reset extruder position
{IF_EXT0}M109 T0 S{TEMP0}
G1 E20 F100
G92 E0 ; Reset extruder position
G1 Z5 F{Z_TRAVEL_SPEED}

the default stop gcode is : 

; Default end code
;G1 X0 Y0 Z130 ; Get extruder out of way. Uncomment to use!
M107 ; Turn off fan
; Disable all extruder
G91 ; Relative positioning
{IF_EXT0}T0
{IF_EXT0}G1 E-1 ; Reduce filament pressure
M104 T0 S0
G90 ; Absolute positioning
G92 E0 ; Reset extruder position
M140 S0 ; Disable heated bed
M84 ; Turn steppers off

nallath, 2019-11-18:
The if statements (eg; {IF_EXT0}) won't work within Cura, so unless the firmware can parse those they won't work.

You can also just try to change the start codes in the machine settings dialog. Could you try if that works (that way we at least know if the start & end g-code you propose will work.

fieldOfView, 2019-11-18:
The other replacement patterns (text between {}) is also incorrect. You have to use eg `{material_print_temperature_layer_0}` instead of `{TEMP0}` for Cura.

For the (unofficial, but complete) list of patterns you can use, see http://files.fieldofview.com/cura/Replacement_Patterns.html

Leleu256, 2019-11-19:
> You can also just try to change the start codes in the machine settings dialog. Could you try if that works (that way we at least know if the start & end g-code you propose will work.

Unfortunately, I could not test the GCODE as the interface do not let me see/modify the START or END GCODE. There must be a bug in the display or maybe because the provided gcode is not valid.
Picture attached.
[image]

I will try with an upgraded Cura 4.4

Leleu256, 2019-11-19:
> I will try with an upgraded Cura 4.4
I'm dumb... The window needed to be expanded to show the content... :-( 

I will try with theses custo GCODES and let you know...

START : 
```
M104 T0 S{material_print_temperature_layer_0}
G28 ; Home extruder
G90 ; Absolute positioning
M82 ; Extruder in absolute mode
G92 E0 ; Reset extruder position
M109 T0 S{material_print_temperature_layer_0}
G1 E20 F100
G92 E0 ; Reset extruder position
```
STOP
```
; Default end code
G1 X0 Y0 Z130 ; Get extruder out of way. Uncomment to use!
M107 ; Turn off fan
; Disable all extruder
G91 ; Relative positioning
T0
G1 E-1 ; Reduce filament pressure
M104 T0 S0
G90 ; Absolute positioning
G92 E0 ; Reset extruder position
M140 S0 ; Disable heated bed
M84 ; Turn steppers off
```

nallath, 2019-11-20:
You can drag the window so it re-scales (and you can see the start & end g-code)

commit 961cd937 (2019-12-04): Correct start and end g-code

According to the thread at #6673 this works.

Fixes #6673.
closed 2019-12-04 by Ghostkeeper