Ultimaker/Cura #6674: Miscalculation of brim on a large STL

bill-orange, 2019-11-18:
Application version: 4.3

Platform: Windows 10

Printer: Anet A8

Reproduction steps: load the  STL from the link below and reduce size to 18%.  Rotate upright.  Set line width to 0.06 and and slice with BRIM on

Actual results Brim is misplaced by 90 degrees and offset.  Opserve in PREVIEW

Expected results: normal properly located brim

Additional information: It is not necessary to run the gcode.  The misplaced Brim is visible in preview.
Image:

[image]

Link to STL:  https://1drv.ms/u/s!AmXKqAwyCrbxiMQQFix5SArPKt16lQ?e=8XNcfQ

fieldOfView, 2019-11-18:
Just looking at the result, it would be my guess the bottom is not actually 100% flat and the brim is getting drawn around the part of the model that actually touches the buildplate. You can "push" the model say half a mm into the buildplate to make the bottom perfectly flat (or just fix the bottom in a 3d package).

bill-orange, 2019-11-18:
@fieldOfView Good catch!  The owner of the OBJ file put there license statement on the bottom apparently screwing up the flatness. I had to go down over 0.5mm

bill-orange, 2019-11-18:
Closiing

closed 2019-11-18 by bill-orange