Ultimaker/Cura #6675: "Could not Probe OpenGL"

dtoxftw, 2019-11-18:
My other PCs/Laptops have never run into this.  But my main "printer machine" does. Cura had been working great for months, and then suddenly I'm getting this error. I found this thread: https://github.com/Ultimaker/Cura/issues/6384 but I can't be sure if this is whats happening and wanted to see if others have found a solution.  Seems strange it would just up and stop and Ultimaker customer support isn't sure either.  They did ask me to update my CPU drivers just to make sure, but unfortunately that didn't help. 

Windows 10 64bit, v1903
i7-4790K (integrated gfx) (15.40.42.5063)
Creality 4.2 and 4.3 both give me this error  :(
Ender 5

Any help is appreciated! 

[image]
Error box: Failed to probe OpenGL. Could not probe OpenGL. This program requires OpenGL 2. or higher. Please check your video card drivers.

nallath, 2019-11-19:
I don't think there is a whole lot that we can do. Your drivers state that you only support a very low version of openGL. You could try to upgrade or downgrade them though.

Ellecross, 2019-11-19:
@dtoxftw you're not accessing that windows PC via remote connection, are you? 
...and do you have multiple display adapters? 
Can you take a look at https://github.com/Ultimaker/Cura/issues/5474 as well? I posted some possible workaround that may be you can try.

dtoxftw, 2019-11-19:
@nallath  the 4790k supports openGL 4.3. The old driver suddenly stopped after working and unfortunately updating it didn't help though.  I'm blaming Windows at this point lol

@Ellecross I was using Windows Remote access!!  I forgot to mention that.  That seems to be the issue. The work around is to simply open Cura locally on that machine and THEN remote in.  

So weird though.  I still don't understand why it would suddenly stop working. Must be the Windows update mentioned in the other thread?  (Before, I had connected to that machine from several different machines in the house, with no Cura issues.) 

I was thinking of getting a dedicated GPU instead of using the integrated but it sounds like that's not a great option.  But in the end, opening/closing Cura locally will work fine for my needs since I can remote in afterwards.

Ellecross, 2019-11-20:
I have always been having that issue (even before Cura 4.0) because sometimes I test Cura remotely as well and I see that myself. 

I hope the other thread workarounds helped you! 

Are you ok to close this issue?

Build-0-Matic, 2019-12-17:
I have been having the same problem but found a temporary fix. It seems that the new version of OpenGL32.dll that is part of Windows is defective. So I ended up finding a version of Mesa that has been compiled for Windows10 X32 and X64. Mesa is a software renderer for OpenGL code. The file consists of a single opengl32.dll file that you put in the Cura folder by hand. While it's not as fast as using a hardware renderer, it does the trick. Here's the link https://fdossena.com/?p=mesa/index.frag

Homeelp, 2020-07-21:
> I have been having the same problem but found a temporary fix. It seems that the new version of OpenGL32.dll that is part of Windows is defective. So I ended up finding a version of Mesa3D that has been compiled for Windows10 X32 and X64. Mesa3D is a software renderer for OpenGL code. The file consists of a single opengl32.dll file that you put in the Cura folder by hand. While it's not as fast as using a hardware renderer, it does the trick. Here's the link https://fdossena.com/?p=mesa/index.frag

Síiiiiiii! Funciona para el Cura 4.6! <3<3<3<3<3

Yeeees! Its works in Cura 4.6! <3<3<3<3<3

Loves [https://fdossena.com/?p=mesa/index.frag](url)

closed 2020-07-23 by Ghostkeeper

touchoftech, 2021-05-03:
> I have been having the same problem but found a temporary fix. It seems that the new version of OpenGL32.dll that is part of Windows is defective. So I ended up finding a version of Mesa3D that has [...]

Thank you for this solution.  I only use remote desktop with windows server 2012 and it works fine with the RTX 480 I put in.

Build-0-Matic, 2021-05-03:
I'm really glad you find it useful. I never thought people would need to run it in remote desktop!

jildertviet, 2022-11-01:
On Linux Mint 21 Cinnamon 5.4.12, Intel i7-1280P, I got the same error.
[This](https://stackoverflow.com/questions/71010343/cannot-load-swrast-and-iris-drivers-in-fedora-35/72200748#72200748) fixed it for me.