Ultimaker/Cura #6677: Prime Tower uses separate X,Y dimensions for each extruder

b-morgan, 2019-11-19:
**Application version**
4.2, 4.3

**Platform**
Windows 10

**Printer**
Custom FFF Printer (LulzBot TAZ 6, Dual Extruder V3)

**Reproduction steps**
Load a model with two colors (3DBenchy)

**Actual results**
The prime tower with dual extruders allocates separate X,Y dimensions for each extruder (concentric squares) but on layers that only use one extruder, the square for the other extruder is not printed. When the second extruder is introduced, there is nothing underneath so the filament is extruded into empty space (i.e. printed in mid-air).

**Expected results**
I believe the way this should work is that for layers that only use one extruder, the prime tower should have both squares printed with that extruder. When a layer uses both extruders, each extruder should print its allocated square in the prime tower.

Ellecross, 2019-11-19:
@b-morgan can you attach a project file, please? 
Also, I'm trying it out now but 4.3 does not have the square prime tower anymore - I cannot reproduce what you're seeing. Neither does 4.4. Can you try on the latest Cura 4.4 if you still see this, please? 

Do you see this only when it's actually printing and not on the gcode?

b-morgan, 2019-11-19:
I'll work on getting a project file and I'll try the latest version.

Yes, I did see it while actually printing. Originally posted this in the [Ultimaker Forums](https://community.ultimaker.com/topic/29218-prime-tower-uses-separate-xy-dimensions-for-each-extruder/) in August using the then current release. It wasn't until recently that I was pointed to Github as the place to post bugs.

nallath, 2019-11-20:
The position of the prime tower is shared between extruders, so I'm not quite sure what you mean.

Ghostkeeper, 2019-12-04:
I think we're looking at expected behaviour here. The expected behaviour of the prime tower is:
* The tower layer for the extruder with the greatest Adhesion Tendency is always printed every layer. This is the one that's on the outside as well.
* The other extruder(s) are only primed if they switched to that extruder on the current layer. So indeed, those extruders will print in mid-air. However since this is done inside of the prime tower it doesn't matter; it will just fall down inside the "purge bucket".

Another way to look at this current behaviour is that we're really just printing a purge bucket with one of the extruders, and with the other extruders we're just purging inside this purge bucket instead of printing a prime tower. It just so happens that we're purging by drawing circles along the edge of the bucket, because that curls up less than if we were to purge in the centre of the bucket.

Let us know if we've misunderstood your problem. Printing in mid-air is not a problem if it's inside the prime tower.

closed 2019-12-04 by Ghostkeeper