Ultimaker/Cura #6679: support retract restrict

lwbsbde, 2019-11-19:
4.3.

support retract restrict do not work

Ellecross, 2019-11-19:
Hey @lwbsbde can you please put more information in your ticket? 
Like, how to reproduce, maybe attach a project file, or some more description... this will help us try to understand what your issue is. 

Thanks

lwbsbde, 2019-11-20:
"Limit support Retraction" does not seem to work.
nevertheless many retractions are made within the support.
What exactly should the function do?

Ghostkeeper, 2019-12-04:
Here is an article in the Settings Guide that describes the expected behaviour of this setting: [Limit Support Retractions](https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/limit_support_retractions.md)

So if Limit Support Retractions is enabled, the filament may still be retracted if:
* Travelling from the model to the support.
* Travelling from the support to the model.
* Travelling from support to support while avoiding collisions with the model (it had to make a detour, not in a straight line).

As far as I can see, this seems to be working correctly. We'll need a project file that reproduces the problem in order to debug this any further.

closed 2020-06-19 by no-response[bot]