Ultimaker/Cura #6680: [4.4.0 BETA] Cannot deactivate auto start print job and webcam image

yet-another-average-joe, 2019-11-19:
**4.4.0 BETA/OctoPrint Connection 3.5.10**
**Windows 10**
**Tevo Tornado/Marlin 2.0**
(only one printer, but 8 were created depending on the nozzle)

**Reproduction steps**
Install 4.4.0 BETA 
Connect a printer (OctoPrint dialog)
Uncheck auto start print job and webcam (I always deactivate these options, and they got activated after installation of 4.4.0 BETA)

**Actual results**
These options are forced "on" by 4.4.0 beta. They can be changed but they are not saved. Closing and reopening Preferences, auto start + webcam are always back.

**Expected results**
The user's choice should be used and saved

fieldOfView, 2019-11-20:
Could you open an issue at the OctoPrintPlugin issue queue instead?
https://github.com/fieldOfView/Cura-OctoPrintPlugin/issues

closed 2019-11-20 by nallath

yet-another-average-joe, 2019-11-20:
Done !

I was thinking of a problem with Cura 4.4.0 beta, as I had another issue after installing it.