Ultimaker/Cura #6682: Printer settings update

andrewjm1959, 2019-11-20:
Issue with version 4.4.0
Printer settings update from previous versions
When updating from 4.0.0 only 3 of my custom printer settings where transferred over.
Version will not let me import new CURA settings for printer.
This issue has happened before and I am currently stuck with using version 4.0.0.
for further information please email.

Ellecross, 2019-11-20:
How do you "import new Cura settings for printer"? 
Can you send me a reproduction scenario so I can try it myself? Thanks

andrewjm1959, 2019-11-20:
Most of the time the settings in the older version of Cura just transfer over. If you want to try to import a set of settings then i have supplied 4 sets from 3Dlabs that are settings for R/C aircraft prints. Theses settings have been imported into version 4.0.0 but 4.4.0 will not let me import them. Standard proceedure is to click IMPORT and then go to file and click on file to be imported.

Ghostkeeper, 2019-12-04:
So currently as I understand, these are your reproduction steps:
* Open profile manager.
* Click the import button.
* Select a .curaprofile file.

And then you get an error message? Which error message are you getting?

When I try these reproduction steps it works fine. A new profile appears in the list.

We need to know from you:
* Which printer are you using?
* Which file are you importing?
* What are you seeing when you import this file?
* What do you expect to see instead?

Ghostkeeper, 2019-12-04:
Also keep in mind that Github doesn't show us any e-mail attachments so please give a URL to the profiles or upload it on the Github website.

andrewjm1959, 2019-12-05:
I have had no problems with importing CURA config settings for 3DLabs configuration files in all version up till 4.4.0. Version 4.0.0. still imports any config files from these model designs which, come preset with G-code files. The models come with G-code files, Factory files, STL files and Config files for Cura and Matter control. If you wish a copy of the cura files, please email me and I will forward a copy of the files to you
Regards Andrew Medlock

andrewjm1959, 2019-12-05:
Attached 2 sets of CURA settings in Zip format from 2 different models.

[CURA_setting.zip](https://github.com/Ultimaker/Cura/files/3927767/CURA_setting.zip)
[CURA_setting.zip](https://github.com/Ultimaker/Cura/files/3927777/CURA_setting.zip)

andrewjm1959, 2019-12-05:
Have also sent ZIP copes to this query . Hope this helps
RegardsAndrew Medlock    On Thursday, 5 December 2019, 11:43:04 pm AWST, Andrew Medlock <andrewjm1959@yahoo.com.au> wrote:  
 
  I have had no problems with importing CURA config settings for 3DLabs configuration files in all version up till 4.4.0. Version 4.0.0. still imports any config files from these model designs which, come preset with G-code files. The models come with G-code files, Factory files, STL files and Config files for Cura and Matter control. If you wish a copy of the cura files, please email me and I will forward a copy of the files to you
Regards Andrew Medlock

Ghostkeeper, 2019-12-06:
Thanks. Those profiles seem to import fine for me, though:

[image]

Which printer are you using?

andrewjm1959, 2019-12-06:
Creality Ender-3

Ghostkeeper, 2019-12-09:
Then indeed you get an error: 
[image]

It would've helped to know the error message when you're reporting a bug!

Ghostkeeper, 2019-12-09:
It seems that our LegacyProfileReader plug-in always uses the quality type `normal` since the legacy profiles obviously didn't have quality types in them.

https://github.com/Ultimaker/Cura/blob/6aa8d060d5ecfb764d5b7ffe15f9415090647ad0/plugins/LegacyProfileReader/LegacyProfileReader.py#L150

Since `normal` doesn't exist for the current printer (Ender 3) it doesn't know what quality type to assign the custom quality to.

andrewjm1959, 2019-12-09:
This is exactly what I get. This is why I am Using the 4.0.0. version as it does not come up with this error message and imports the settings without any trouble. Unfortunately the settings will not transfer Over to the new version , When I install it onto my computer. This comes down to a bug within the program that was not there in 4.0.0..I am also using the new motherboards in both my printers and, they both will not now work with REPETIER HOST.I have reported the issue to them but the have requested  a donation to look at the problem to fix it, very poor PR as i have been using them for over 3 years now without any problems. The 2 motherboards have 2 different versions of MARLIN on them but both run much better then the original. Why software companies cannot get their act together is beyond me.

 
 

https://github.com/Ultimaker/Cura/blob/6aa8d060d5ecfb764d5b7ffe15f9415090647ad0/plugins/LegacyProfileReader/LegacyProfileReader.py#L150

sebachu, 2019-12-10:
I have this exact same problem trying to upgrade from 4.3 to 4.4, with an Ender 3. I've downgraded to 4.3 because I need my custom profiles to import. Please let me know if there is anything I can do to help troubleshoot.

Ghostkeeper, 2019-12-10:
We've decided to defer this, not spend time on it unless time opens up or priority increases. However we've dropped support for Cura 2.6 and earlier so I don't expect that we'll get time to fix this for a long time. Unless someone from the community decides to spend time on it instead. The problem lies in the line that I quoted above. Instead of `normal` it should probably choose one of the available quality types in the current configuration as defined in the container tree.

Our advised workaround is to import all your profiles into 4.0 and then upgrade to 4.4. Two steps instead of one. Alternatively you can import them in 4.0, export them from there and import them in the latest version again.

If you have problems trying to upgrade from 4.3 to 4.4 that is a separate bug and I'd really like to know about it. But you'll have to write a new bug report about it to not confuse the two bugs.

andrewjm1959, 2019-12-10:
Hi Ghostkeeper and, TY for your time.  I thought you might also like to know, that I have 2 Ender3 printers. A new  one with version 1.1.6.2. motherboard and the original which now has 1.1.8. motherboard fitted to it. The new motherboards work with major improvements in print quality but, they both do not work on Repeteir Host. Repeteir are looking into the matter and I intend to get onto Creality about the issue. It is obvious that somewhere along the line, when they upgraded the Marlin software they altered it so it would not work with Repeteir. I hjave also had troubles with them working with CURA but am now just using the SD cards for all my printing. Creality have put their own slicer program with the printer software but, I am not impressed with it and am sticking to Cura for my slicing needs. TY again for your time and help
Andrew Medlock

xb4r7x, 2019-12-26:
Same problem. It stinks that it won't be fixed.

Ghostkeeper, 2019-12-30:
Did you try importing the profiles into Cura 4.0 and then upgrading them to 4.4?

andrewjm1959, 2019-12-30:
Hi, Merry Xmas and Happy New Year. Yep, that's what I did. Files transferred fine to 4.0 but only 1 set transferred to 4.4. Just gonna stick with 4.0 until the next update comes out and then try and transfer them over to that.

xb4r7x, 2019-12-30:
> Did you try importing the profiles into Cura 4.0 and then upgrading them to 4.4?

I did not. I ended up determining the necessary settings on my own and changing them manually. Pain in the butt, but I got my prints going.

andrewjm1959, 2019-12-31:
I guess it would have been...LOL. I have 27 profiles in my custom file settings now and, I don't fancy manually transferring them in else I would have a major headache.Thinking of writing the whole lot down as some are minor changes for set jobs. Had over 50 when I was using Repetier Host which had 3 slicer programs in it.

ComputerKing12, 2020-06-04:
There is a very simple way to fix this, navigate to Cura's installation directory, then plugins, then navigate into the LegacyProfileReader plugin folder. Open LegacyProfileReader.py in a text editor, and change line 150 from normal to super

LeoEckert, 2021-01-17:
> There is a very simple way to fix this, navigate to Cura's installation directory, then plugins, then navigate into the LegacyProfileReader plugin folder. Open LegacyProfileReader.py in a text editor, and change line 150 from normal to super

Does that require an installation of a certain plugin? I am facing the same issue importing a profile from a guy with an ender 3 to my custom printer. When I navigate to the plugins directory, it is empty here.

Ghostkeeper, 2021-01-25:
> Does that require an installation of a certain plugin? I am facing the same issue importing a profile from a guy with an ender 3 to my custom printer. When I navigate to the plugins directory, it is empty here.

No, this was talking about modifying the built-in plug-ins that Cura ships with. They are probably located in `C:\Program Files\Ultimaker Cura 4.8\plugins`, not in `C:\Users\<username>\AppData\Roaming\cura\plugins`. So the installation folder, not the configuration folder. On Mac and Linux, you'd open up the application with an ISO disc image reader and modify it in there (but then on Mac the signing will fail).

andrewjm1959, 2021-01-25:
TY
Go it all sorted. As  i am using an ENDER3, I did a search and came up with this web site which fixed all my problems.
https://www.youtube.com/watch?v=Ztygn--BsjI&t=207s

RegardsAndrew Medlock

eboutin, 2021-07-24:
FYI I successfully imported 4 profiles from 3dlabprint.com (for the Qtrainer model) into Cura 4.10 using instructions from @ComputerKing12.

I used VS code to modify C:\Program Files\Ultimaker Cura 4.10.0\plugins\LegacyProfileReader\LegacyProfileReader.py.

I modified line 159 from:
        output_parser["metadata"]["quality_type"] = "normal" # Don't know what quality_type it is based on, so use "normal" by default.
to:
        output_parser["metadata"]["quality_type"] = "super" # Don't know what quality_type it is based on, so use "normal" by default.

Whatever file editor you're using for this operation it has to be executed in Administrator mode. VS code does this automatically for you.
The modification must be performed in Admin mode

Ghostkeeper, 2021-07-30:
That change would cause it to break for other printers again.

A better change would be to change it to whatever the active quality_type is or the preferred quality_type of the printer.

PBSDMark, 2022-01-13:
Hi Guys,

I’ve been trying to use the Cura profiles from 3dLabPrint for my P47 and it was failing on import for the ENDER 3.  It was looking for a quality type called “normal” which doesn’t exist for the Ender 3.

Here’s the fix:

You need to modify only of Cura’s plugin files called LegacyProfileReader.py 

They are probably located in C:\Program Files\Ultimaker Cura 4.8\plugins, not in C:\Users\<username>\AppData\Roaming\cura\plugins

•	Open it in Notepad.

•	Find the line #159: 
•	output_parser["metadata"]["quality_type"] = "normal" # Don't know what quality_type it is based on, so use "normal" by default.

•	Copy the line and paste it right below the orignal line

•	Then put a pound-sign “#” in front of the ORIGINAL LINE to comment it out (it will be ignored), but lets save the original “just in case”.

•	CHANGE the word “normal” to “super”

•	Save the file

•	Restart Cura

•	Import your Cura Slicer files now with no issues

Ghostkeeper, 2022-01-20:
Of course, that would give problems importing profiles when the active printer doesn't have a quality type "super".

magginichi, 2022-10-06:
If somebody else still has that problem. This fixed it for me:

Just change the quality_type from normal to standard inside the profile:
`quality_type = standard`

1. Unzip all files inside .curaprofile file to a folder
2. Edit all files with an editor (like vscode) and change `quality_type = normal` to `quality_type = standard`
3. Save your work
4. Zip the files again and rename it with .curoprofile instead of .zip or replace the original files inside the .curoprofile file

closed 2024-12-07 by github-actions[bot]