Ultimaker/Cura #6683: Cura 4.4.0 Not Slicing

BigDangler55, 2019-11-20:
**Application version**
(The version of the application this issue occurs with.)
4.4.0
**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)
Windows 10
**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)
Ender 3 Pro
**Reproduction steps**
1. Something you did.
I downloaded the most current update for Cura. Then, I opened an .stl file into cura and attempted to slice the piece. The slice button does not seem to work.
2. Something you did next.
I then went from the prepare page to the preview page and attempted to slice again.  Still didnt work so I closed and reopened the part file.  Still didnt work so I attempted a different file I have already printed in the past. Still didnt work so I closed and reopened Cura.  All of these steps did not help cura to slice. when slice button is pressed.
**Actual results**
(What happens after the above steps have been followed.)
Absolutely nothing.  Part stays greyed out because the slicing did not occur
**Expected results**
(What should happen after the above steps have been followed.)
Part should turn blue and yellow with sliced layers so that I can export to an SD card.
**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)
No interface feedback occurs when attempting to click the blue slice button.  This is the first time Ive ever had this issue with Cura and I just downloaded the new update Cura 4.4.0 on 11/20/19

rburema, 2019-11-21:
- What do the logs say? They're in `%appdata%/cura/4.4/cura.log`
 - Could you please send us one of the project files that didn't work for you?

Goodfeat, 2019-11-21:
Slicing will work if you install a clean version without settings. it helped me. most likely this is a conflict of settings of the previous version

rburema, 2019-11-21:
> Slicing will work if you install a clean version without settings. it helped me. most likely this is a conflict of settings of the previous version

The upgrade-scripts should've taken care of that, though ...

Goodfeat, 2019-11-22:
it seems the reason why the program cannot do slicing is the custom file "fdmprinter.def.json"
The file I use is different from the original.
I tried moving part of the code to custom file.
[image]
slicing was successful

rburema, 2019-11-22:
I see. So your problem is probably different from the original post.

BigDangler55, 2019-11-23:
Thank you for your suggestion.  After I left Cura open on my computer for a few hours, yesterday the button finally allowed me to press it. It now seems to be working properly. Thanks again

rburema, 2019-11-25:
I'll close this now as the problem was solved :-)

closed 2019-11-25 by rburema