Ultimaker/Cura #6684: Laptop touchpad two-finger scrolling on certain dialogs is very hard to control

ColonelThirtyTwo, 2019-11-20:
**Application version**
4.2.0, from pacman (Arch Linux) and AppImage (Ubuntu)

**Platform**
Happens on both Arch Linux with Gnome and Ubuntu 18.04.3.

**Printer**
Not Applicable

**Reproduction steps**
1. Using a laptop touchpad, use two fingers to scroll certain scrollable dialogs, such as the What's New and Add Non-Networked Printer dialogs. The print settings dropdown is NOT affected.

**Actual results**
The dialog continues to scroll with leftover velocity after the user stops moving their fingers or removes their fingers from the touch pad.

**Expected results**
The dialog stops scrolling when the user stops moving their fingers, stopping at where the user intends is to stop.

**Additional information**
This issue make scrolling the affected dialogs via two-finger scrolling basically impossible.

Honestly, I'd recommend just removing the weighted scrolling. It's not a useful feature for desktop, and as this issue demonstrates, it's not implemented very well.

nallath, 2019-11-22:
It was a quick fix to change this behavior for the machine selection, but it's a bit more tricky to disable it for the what's new dialog (the flicking behavior isn't something we put in there, it's a default thing that the UI framework that we use does)

ColonelThirtyTwo, 2019-11-26:
Also happens with all of the tabs in the marketplace window

Ghostkeeper, 2019-12-04:
This is actually just a Qt bug: https://bugreports.qt.io/browse/QTBUG-56075

I think the printer list scrolling is now too slow with a scroll wheel.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

oliof, 2024-11-22:
I tested on the list of printers to add from OP. While there is still a slight over scroll effect if you end the scrolling by taking both fingers off the touchpad, if you keep the fingers on the touchpad precise targeting of the scroll end position is possible.

I am a heavy touchpad usernamd would say this isnt a problem anymore (other than the scrolling speed limit on the settings pane but I believe thats independent and works as implemented).

GregValiant, 2024-11-22:
Thanks.  I'll close it.

closed 2024-11-22 by GregValiant