Ultimaker/Cura #6685: Drops UM Original nozzle size in to default

alexshorokhov, 2019-11-20:
Cura 4.2-4.4 do not save nozzle size changes for UMO in machine settings.
I change default 0,4 to my 0,5, and after closing a window nozzle size returns to 0,4.
[image]

rburema, 2019-11-21:
Can't seem to reproduce this in 4.4 (even if I try it with a `,` instead of a '.') It stays the same even after restart.

Perhaps it's (still) related to locale settings though, could you  try it with a '.'?

Alternatively, is this a machine that has been there before the upgrade? Does this still happen when adding a 'new' UMO?

alexshorokhov, 2019-11-21:
It still happens when adding a new UMO. No matter if it is dott or comma, no result.

rburema, 2019-11-21:
What's your locale? Maybe I can reproduce it that way.

alexshorokhov, 2019-11-29:
What do you mean "locale"? English is not my common language, so I can misunderstand you.

rburema, 2019-12-03:
A 'locale' in the sense that I meant is basically the (collection of) internationalisation settings that (a program or) the OS is set to. For instance, if I set my locale to Dutch (nl_NL), then, not only will the language change, but the default decimal separator will be a `,` while the thousands separator will be a `.` (the opposite of what it is in English (en_EN)).

alexshorokhov, 2019-12-03:
Locale is Russia. Cura language is english.
I've just done it! I've typed ",5" when the keyboard language was English instead of Russian and pressed enter. Then Cura automatically changed " , " to ". " and saved this setting. Now it is 0.5.
Many thanks!

alexshorokhov, 2019-12-03:
I think, I will close tthis issue now)

rburema, 2019-12-03:
You're welcome!

I'm not sure if it _should_ be closed though... it should still work after all, even if your language isn't English.

Ghostkeeper, 2019-12-05:
No, we'd like to keep this open because Cura should also work properly in the Russian locale.

alexshorokhov, 2019-12-05:
Ok.

Ellecross, 2019-12-05:
Ok - this is weird. I have tried it myself, change my locale to Russia, had a Russian language keyboard - but had to switch to english when change the Nozzle size setting. 

I also switched to Russian keyboard and Cura converts 0,5 to 0.5 automatically. 

So, I cannot reproduce it. :/ The values always get saved.

Ellecross, 2019-12-05:
> No, we'd like to keep this open because Cura should also work properly in the Russian locale.

I don't know, I think we should close it and keep an eye out for this in the future.

Ghostkeeper, 2019-12-06:
Hmm, maybe it's different per platform?

DenKoren, 2019-12-08:
I just faced the same problem on Mac OS X with Russian system locale.

When I tried to change nozzle size Cura siletly ignored my changes until I found this thread and tried to use comma instead of dot as a separator.

I think, it would be great to detect system locale and show the values using the default separator for current system locale, if it affects the UI. This will give users a clue about the symbol they should use.

Ghostkeeper, 2019-12-09:
Our idea so far is that the user can type any radix and all of them should work. However they will all be converted to period as to keep things simple and consistent.

DenKoren, 2019-12-09:
Shure, it would be much more convenient. 

My idea was about the quickfix to be considered before the normal patch. It may be easier and faster than locale-independent solution.

Nemori27, 2019-12-10:
Have the same issue, cand change nozzle diameter, whatever i do its still the same. After reading this page i tried to type 0.2 in russian keyboard and it seems to work! Howewer its strange, as my default keyboard language is english and i tiped values in english when it refused to work

closed 2024-12-07 by github-actions[bot]