Ultimaker/Cura #6686: Print objects one after the other

lwbsbde, 2019-11-20:
If I slice in Cura 4.2 several objects in succession, then Cura starts from X0 to X300 .
But Cura 4.4 slice with a different strategy.
I would rather slicing it like Cura 4.2 , because it allows me to print higher parts. :-)

rburema, 2019-11-21:
By 'several objects in succession', so you mean 'one at a time' mode?

lwbsbde, 2019-11-21:
Yes

coofercat, 2019-11-27:
This one?

[image]

coofercat, 2019-11-27:
This one?

[image]

(or am I missing something...?

lwbsbde, 2019-11-27:
How can I set the order of the print at Cura4.4?
Cura 4.2. has been sorted from X0 to X300.

coofercat, 2019-11-27:
That I don't know - In a quick check I just performed, it seems to print in height order, starting with the object furthest from the origin (which it puts there when I say "Arrange all objects")

It's not a mode I use much any more - the available height is reduced so much that it's not very useful.

Ghostkeeper, 2019-12-04:
When printing one at a time, the order in which objects are printed is constrained to their collision areas and the collision area of the print head's shape. If the nozzle is in the front of the print head, they should get ordered from front to back. If the nozzle is in the back of the print head, they should get ordered from back to front. If the nozzle is on the left side of the print head, they should get ordered from left to right. If the nozzle is on the right side of the print head, they should get ordered from right to left.

The collision area of the objects on the build plate is adjusted so that you can never place objects such that the print head collides with the object.

This depends on the print head to be correctly defined in the printer definition, of course. For Custom FFF Printer you'd need to define the size of the print head in the Machine Settings dialogue.

There is no manual ordering of the objects. That would involve creating different collision areas for each pair of objects on your build plate which is complex for the user to understand.

closed 2019-12-04 by Ghostkeeper