Ultimaker/Cura #6687: [4.4.0] Raft doesn't slice correctly

tkoft, 2019-11-21:
3mf file (with fake .zip extension, not real zip):
[shaft_cura_bug.3mf.zip](https://github.com/Ultimaker/Cura/files/3872399/shaft_cura_bug.3mf.zip)

**Application version**
4.4.0

**Platform**
MacOS Mojave 10.14.6

**Printer**
Monoprice Mini Delta

**Reproduction steps**
1. Slice this file with a raft

**Actual results**
Raft doesn't extend beneath all the parts, specifically the horizontally laid shafts

**Expected results**
Raft should extend past base of all parts.

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

rburema, 2019-11-21:
Thanks for the file!

The shafts are not level with the build-plate. Is that correct?

Also, I can only import the models from that file, as I'm missing the printer definition.

tkoft, 2019-11-22:
Ah, it wasn't level.  Not sure how that happened.  I even have snap rotation on, so things only rotate in 15 deg increments...

Resolved, I guess!

closed 2019-11-22 by tkoft