Ultimaker/Cura #6688: two or more extruders are not created for delta

Goodfeat, 2019-11-21:
Hello. I encountered the error of slicing parts in Cura 4.4. (the program could not finish the cut after the start).
I decided that it was a profile issue from previous versions.

The next step I tried to create a new profile for delta
[image]

However, when I went into the printer settings, I could not put two extruders.
[image]

Windows 10
Cura 4.4
Delta printer
nozzle 0.4
dual extruder (extr 1 + extr.support)

Goodfeat, 2019-11-21:
I solved the problem. It turns out that the profile of the delta was not originally intended to be printed by two extruders.

The solution was simple. I added a new line in to the file "deltabot.def.json"
"1": "deltabot_extruder_1"

[image]

I also had to create an additional file deltabot_extruder_1.def.json

it was not implemented initially due to bugs?

rburema, 2019-11-21:
I'll ask around if there was a reason for that.

rburema, 2019-11-22:
Originally, deltabot wasn't in custom, since 'Custom FFF Printer' can also handle this use case. The recommended way to do this (for now) is to use 'Custom FFF Printer', where you _can_ select multiple extruders.

Pull requests are welcome, though!

Goodfeat, 2019-11-22:
I tried the method described above. It seems to work.

Ghostkeeper, 2019-12-03:
> I'll ask around if there was a reason for that.

History. Deltabot was made as an easier way to create a custom Delta printer. It's in fact older than Custom FFF Printer and was implemented before we had multi-extrusion support.

fieldOfView, 2019-12-05:
I don't know why Deltabot ended up in the "Custom" category. Having it there only confuses matters. The definition is a legacy from way back:
https://github.com/daid/LegacyCura/blob/SteamEngine/resources/machine_profiles/DeltaBot.ini

Goodfeat, 2019-12-05:
Indeed, this is confusing.
This option offers no benefits other than the selected circle-shaped bed.
... and only one extruder

Ghostkeeper, 2019-12-06:
Yeah we could try removing it but that involves deleting a bunch of profiles from the users' configuration as well.

fieldOfView, 2020-03-23:
Pull request is here: #7331

closed 2023-06-13 by github-actions[bot]