Ultimaker/Cura #6689: [4.4] Comments removed from Start/End G-codes after update

Block137, 2019-11-21:
**Application version**
4.4 after updating from 4.3

**Platform**
Win10

**Printer**
Custom built

**Reproduction steps**
1. Put some G-code descriptions as new line comments in Start/End G-codes in 4.3
2. Update from 4.3 to 4.4 (uninstalled 4.3 beforehand, kept configuration files)
3. Open machine settings window again

**Actual results**
MOST comment only lines (lines starting with ";") were removed. Comments following actual G-codes ("G28 ;home" for example) remain unchanged.
Commented first line didn't get removed.

No more change between restarts, only once after update.

**Expected results**
Whole G-code scripts remain unchanged.

**Additional information**
https://github.com/Ultimaker/Cura/issues/1237 fixed this issue before, only difference is this doesn't happen between restarts.

My End G-code is as follows:
;<END>
G0 Z305 F1500 ; lower bed
M140 S0
M104 S0 ; M104 T0 S0
M124 S0 ; M104 T1 S0
M134 S0 ; M104 T2 S0
M107
;<DOCK>
T0
G28 X0 
G28 Y0
M84
;</END>

Only ";<DOCK>" and "</END>" got removed. "<END>" at the top remained.

;<DOCK> and other brackets in comments are used by my external post processing script as fix for #5861 and other patches for Smoothieware.

rburema, 2019-11-21:
I can reproduce this. It seems the problem is the upgrade itself as opposed to the script, because if I go into the machine settings in 4.4, the code is reduced there as well. However, if I then put it back and restart (in 4.4), it _does_ remember.

fieldOfView, 2019-11-21:
If the UpgradeManager parses the instance container ini file and writes it back to disk, it should disable comments like here:
https://github.com/Ultimaker/Uranium/pull/301/files

rburema, 2019-11-22:
Traceback: We've made (internal) ticket CURA-6994 for this.

P.S.: @DonExo this is how we can track it back to 'our' system, so we don't spend ages looking if a ticket has already been made (like you just did) from this internally :-)

closed 2019-12-04 by ninovanhooff

Block137, 2020-03-19:
@rburema I just discovered this problem again in 4.5 after opening .3mf file.

The .3mf was generated in 4.5 and opened as project in 4.5.

Should I make a new bug report?

rburema, 2020-03-19:
@Block137 Yes please. As CURA-6994 has been in the done column for a while, so this might be a new (but probably related) issue.

PR #6751 (2019-12-04): CURA-6994/Prevent deleting comments from Start/End G-codes
Prevent deleting comments from Start/End G-codes by changing the `comment_prefixes` in ConfigParser" -m "CURA/6994".

Note: We should add this in every future VersionUpgrade.

Fixes #6689