Ultimaker/Cura #6691: Support stream does not work ( cura 4.3-4.4)

Goodfeat, 2019-11-21:
Hello. I noticed that the support flow  setting does not work.
[image]
[image]

after changing the setting, nothing changes, the plastic consumption remains the same

Windows 10
Cura 4.3-4.4
Delta printer
nozzle 0.4
dual extruder (extr 1 + extr.support)

rburema, 2019-11-22:
You're right. I can't find `support_material_flow` only once in the settings (nothing else refers to it) and not in the engine at all!

rburema, 2019-11-22:
I've pushed a fix for this, should be in next release unless reverted.

(See https://github.com/Ultimaker/CuraEngine/commit/e40ac1c6c322a464805d49c3ddd450bcec54e740 )

closed 2019-11-22 by rburema