Ultimaker/Cura #6692: View -Zoom,move

Bajotaz, 2019-11-21:
Bring back Cura 4.3 mouse movement, the Cura 4.4 mouse movements is FUBAR!
Can not turn an object without loosing object from center of screen...

fieldOfView, 2019-11-21:
AFAIK, Nothing changed wrt zooming and moving between 4.3 and 4.4. 
Have you checked the viewport behavior options in the preferences?

nallath, 2019-11-21:
We also need a bit more information about what it is that you're doing (and what it is that is going wrong). Could you make a (short) video that shows the problem?

Bajotaz, 2019-11-22:
Hi.

I was unable to cirkle around a object.
But it works now.
Had to enable all options in viewport behavior, restart cura, disable all,
restart cura then enable my choices and it worked.

Maybe something with porting saved options to next version?

Ghostkeeper, 2019-12-04:
Maybe your camera dolly position ended up weird and a restart fixed the problem.

closed 2020-06-19 by no-response[bot]