Ultimaker/Cura #6695: The camera button sometimes does not respond when clicking on it.

jficaraharvard, 2019-11-21:
**Application version**
4.4.0

**Platform**
Windows 10 Professional build 1909
GeForce GTX 1070
Driver Vesion 441.12

**Printer**
Ultimaker s5

**Reproduction steps**
1.Opened Cura 4.4.0
2.Navigated to Monitor tab
3.Clicked on the camera icon

**Actual Results**
The camera view did not show up. There was no UI response to clicking on the camera.
The status bar is indicating 87% complete and the UI shows as connected to the printer, displaying the correct printer name, correct filament in extruder 1 and 2.  No project file was open, just opened Cura to check on the status of the printer.

**Expected results**
The camera view should be displayed.

**Additional information**
If I close the app and launch it again I can then access the camera view but sometimes it takes several tries of shutting down the app and restarting to get the camera view to work. Some times it works the first time.

The iOS app is able to access the camera just find even when the desktop cura app mentioned can't. They are both on the same network.

rburema, 2019-11-22:
There might be something in the logs. Can you post `%appdata%\cura\4.4\cura.log`?

jficaraharvard, 2019-11-22:
Here is the log, thank you for looking into this.

[cura.log](https://github.com/Ultimaker/Cura/files/3880417/cura.log)

rburema, 2019-11-25:
Hm, I could find this:
`2019-11-20 17:00:18,852 - WARNING - [MainThread] cura.OAuth2.AuthorizationHelpers.parseJWT [105]: Could not retrieve token data from auth server: {"errors": [{"id": "450a5307-0d7e-4f82-8ee1-11c64fe35b0b", "code": "tokenExpired", "http_status": "403", "title": "The given token has expired.", "meta": {"error_type": "AuthError", "url": "https://account.ultimaker.com/check-token"}}]}`

@ChrisTerBeke could you help me out with this? I seem to remember discussing something related to this with the team and you in a Cura & Cookies session.

Ghostkeeper, 2019-12-04:
Maybe the clock in the printer is drifted too far?

ChrisTerBeke, 2019-12-04:
@rburema the webcam view doesn’t use cloud, so this error is unrelated. Your log line indicates that cura didn’t refresh the cloud access token quickly enough, but that shouldn’t cause the original reported issue.

GregValiant, 2024-11-14:
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

jficaraharvard, 2024-11-14:
Yes this issue is resolved

Best Regards,

Joseph Ficara

***@***.***

GregValiant, 2024-11-14:
Nice.  I'll go ahead and close this.