Ultimaker/Cura #6696: Import does not carry over collapsed items

FlailAway, 2019-11-21:
4.3.0. and 4.4.0, Mint 19.3+Cinnamon
Feature request if it is not already instantiated, but not found by me, yet.

In 4.3.0, I just spent a bunch of time collapsing all of the unneeded (by me) items displayed for a Profile, then Saved and Exported them.

I downloaded 4.4.0 and imported on of those the Profiles and all of the collapsed items were expanded again. I can find "Show all" but nothing about "Show what I want."

Is there a setting for this I have not yet found, or must I go through and collapse all that visual clutter -- again?

rburema, 2019-11-22:
You can manage the settings visibility (last item in the menu), but that'll be saved to your overall preferences, not to the profile. Neither is the collapsed/uncollapsed state of the setting-category headers.

I'll discuss this with my colleagues.

rburema, 2019-11-22:
It seems this would require a overhaul of the entire profile system. As such, we'll have to defer this for now.

FlailAway, 2019-11-22:
> It seems this would require a overhaul of the entire profile system. As such, we'll have to defer this for now.

OK, I really like Cura and I believe it is streets ahead of any other Slicer (free or $150) on the market, but -- and ain't there always a "but," the **overall** user interface is pretty awful. I have been programming for 40+ years and I stand in awe of what you guys are doing technically. Especially since 4.2.0 and 4.3.0, great leaps under the hood.

I'd really like to help with that as the User-Interface was/is my forte and constant nagathon. ;)

Is the profile system compartmentalized or randomly integrated with the main code base?

nallath, 2019-11-25:
The profiles only store values (which is why a pretty major refactor would be needed). The visibility data is stored in project files though.

Anyway, the settings are a relatively different system. So as with any system, we try to keep it compartmentalized, but as time goes, it's always a fight against entropy to keep it like that.

If you have any specific questions wrt to helping out with the code, feel free to ask them. We could really use some more contributors :)

Ghostkeeper, 2019-12-04:
An alternative way of working is provided by fieldOfView's plug-in in the Market Place called "Setting Visibility Set Creator". It's not stored along with the exported quality profiles, but you can save the setting visibility sets at least and even copy/paste them from the configuration folder if you ever need to move them (or use the Back-up feature with the Ultimaker accounts).

I think that's an acceptable workflow for now and we won't be implementing a way to export setting visibility sets via the quality profile exports.

closed 2019-12-04 by Ghostkeeper