Ultimaker/Cura #6697: Auto export from AutoCAD Fusion 360 to Cura 4.4 not working

luisguevarap, 2019-11-21:
**Application version**
4.4.0

**Platform**
Windows 10
**Printer**
Creality 10S

**Reproduction steps**
1. Using model export feature in AutoCAD Fusion360 to export to Cura
2. Until version 4.4 the export feature in AutoCAD Fusion360 would automatically open Cura and automatically open the file ready for positioning and slicing. 

**Actual results**
Now nothing happens. I have to export and .STL file to folder then open Cura and open the file. A lot of extra steps. 

**Expected results**
Should automatically open Cura and automatically load the file into Cura

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

rburema, 2019-11-22:
It seems to me that its AutoCAD's repsonsibility to open Cura. Cura can't open itself when it's not running after all.

There might still be something we'd be able to do, but the search has to start on their side of things. Especially since Cura still opens with the right model when I give it the `.stl` as an argument from the command line.

luisguevarap, 2019-11-23:
Thank you. 

I found this from a previous change in version. I have notified Autodesk Fusion 360 folks. 

 

Fusion360 attempts to find Cura by looking in the registry at

   HKEY_CLASSES_ROOT\Cura STL model file\shell\open\command

 

That's not where Cura 3.1 stores its registry information.   The key is now called

   HKEY_CLASSES_ROOT\Cura.model.STL\shell\open\command

 

You can see this in their installer script at:  <https://github.com/Ultimaker/cura-build/blob/48a17f307a65f558674a1e1ab5c4bb977bfc129f/cmake/NSIS.template.in#L548-L554> https://github.com/Ultimaker/cura-build/blob/48a17f307a65f558674a1e1ab5c4bb977bfc129f/cmake/NSIS.tem...

 

By default, this key does not have the --single-instance flag.   Attempting to add the flag to the registry does not help.   If it is after the "%1" value, Fusion360 will ignore it and call Cura without the flag.   If it is before the "%1" flag, Fusion360 will fail to launch Cura.

 

In short, Fusion360 needs to change its Cura integration code to

1.) Obtain the location of Cura from the registry key used by current versions (Cura.model.STL instead of Cura STL model file)

2.) Append " --single-instance" to the value retrieved from the registry when invoking Cura.

 

That should be an easy fix for the Autodesk folks to make. 

 

 

Luis Watts

rburema, 2019-11-25:
Thanks for putting in some legwork of your own! I don't know them, but I know I'm always happy when I get a report that clear.

(Closing this now, as we can't do anything about it.)

closed 2019-11-25 by rburema

Matioupi, 2020-01-03:
Just as a follow up, I posted a report on Fusion 360 forum about this buf that I'm also facing :

https://forums.autodesk.com/t5/fusion-360-support/fusion-360-not-starting-cura-4-4/td-p/9229529

jdzawodniak, 2020-02-28:
A work-around: 
One can use the Fusion 360 - 3D Print - Print Utility "Custom" function to automate the launch of Cura.  Simply point the Application dialog to the Cura.exe (C:\Program Files\Ultimaker Cura 4.4\cura.exe)

Then select the body as usual and hit ok.