Ultimaker/Cura #6698: Request to add a limit function for adaptive layers

Goodfeat, 2019-11-22:
Hello

Is it possible in the new version of the program to implement the function of limiting the adaptive layer to the maximum layer

[image]

The user has to constantly change the layer height if the adaptive layer mode is activated
This mode allows you to limit the variability of the layer setting for the upper limit of "Layer Height"

nallath, 2019-11-26:
But if you want the layers to vary between 0.28 and 0.48, you could also set the layer height at 0.38 and the max variation at 0.1 right?

Goodfeat, 2019-11-27:
Yes, it is true. But unfortunately, you have to manually calculate the height parameter each time. If it is not calculated, then this parameter goes beyond the nozzle size limit. I have to control both parameters every time

I use the adaptive layers option as a frequent tool. I had to modify fdmprinter.def.json so that the parameter automatically calculated when adaptive layers were turned on.
This is suitable for me as a workaround. But this solution has obvious disadvantages.
In addition, the internal algorithms of the program for some reason do not round the numbers in the calculation.
Instead of 0.32, the program displays the number 0.3199999999 ...
I had to select formulas so that the program displayed the correct numbers.
Here is my current code so that the program calculates the values ​​when the adaptive layer is turned on

```
                "layer_height":
                {
                    "label": "Layer Height",
                    "description": "The height of each layer in mm. Higher values produce faster prints in lower resolution, lower values produce slower prints in higher resolution.",
                    "unit": "mm (calc.-23%)",
                    "type": "float",
                    "default_value": 0.31,
                    "value": "(machine_nozzle_size - (machine_nozzle_size*0.20))- (adaptive_layer_height_variation) if adaptive_layer_height_enabled else (machine_nozzle_size) - (machine_nozzle_size*0.20+0.02)",
                    "minimum_value": "0.01",
                    "minimum_value_warning": "0.04",
                    "maximum_value_warning": "0.81 * min(extruderValues('machine_nozzle_size'))",
                    "settable_per_mesh": false,
                    "settable_per_extruder": false
                },
```

Please add this function in a future version of the program, This function simply adds restrictions, theoretically the function should not break the program

Ghostkeeper, 2019-12-04:
Perhaps it's more clear to have a "maximum layer height" and a "minimum layer height" setting when adaptive layers is enabled, and then disable the ordinary layer height setting?

Goodfeat, 2019-12-05:
That would be great.
your option looks much more convenient

in any case, setting the upper limit limit is very necessary

Ghostkeeper, 2019-12-17:
We've decided to reject this. Changing to a "max" and "min" is not feasible since we'd have to refactor many places in CuraEngine that expect to have a layer height. You'll need to calculate the average of the min and max yourself and then select the appropriate variation that you want.

closed 2019-12-17 by Ghostkeeper