Ultimaker/Cura #6700: Can not start Cura 4.4 on Mac

EL250a, 2019-11-22:
**Application version**
Cura 4.4
Mac

**Printer**
Cr10spro

**Reproduction steps**
Open Cura 4.4

**Actual results**
"Ultimaker Cura" cannot be opened because Apple cannot check for malicious software.

**Expected results**
Open the program

**Additional information**

[image]

sparky14, 2019-11-23:
I am having the same issue on Mac Catalina 10.15.1

smartavionics, 2019-11-23:
You could try one of my builds that can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. They have been built using a more recent version of QT compared to the UM releases. Please read the README.md file there. They can be installed alongside the UM builds without conflict.

Ghostkeeper, 2019-12-04:
Though the symptom is different, we're merging this as a duplicate of #6643. The problem is the same. We're currently using an old version of Qt in order to support MacOS 10.11. Upgrading the Qt version would drop support for MacOS 10.11 but apparently gain support for 10.15. There is no option to support both. So I think we'll choose 10.15 over 10.11 then.

closed 2019-12-04 by Ghostkeeper