Ultimaker/Cura #6702: Feature Infill Line Multiplier conflict with Randomize Infill Start and leads to crush.

digiTALker, 2019-11-23:
**Application version**
Cura 4.4.0

**Platform**
Win 7. Intel, Nvidia

**Printer**
Custom printer Priusa kinematic. Repitier.

**Reproduction steps**
1. Check box "Randomize Infill Start"
2. Set Infill Line Multiplier to "2"

**Actual results**
Leads to crush some function of Cura. But after closing error message Cura still works. Looks like crash in code generating module.

**Expected results**
Expected correct end of slising. 

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)
[https://i.imgur.com/9BtTN87.png](url)

digiTALker, 2019-11-23:
https://i.imgur.com/9BtTN87.png

smartavionics, 2019-11-23:
Which infill pattern are you using? I just tried those settings and it worked OK. Please save the project file (File -> Save) and zip or rename the .3mf file and attach to this issue, thanks.

Ghostkeeper, 2019-12-04:
I can reproduce this problem. Here is a project file inside a zip file:
[UM3_tetrahedron.zip](https://github.com/Ultimaker/Cura/files/3922421/UM3_tetrahedron.zip)

DkOuvE, 2020-02-14:
Can also confirm this bug. When I turn off randomise, it slices fine.

Ghostkeeper, 2020-02-18:
We discussed this a long time ago, but I don't know what happened with the ticket.

I've made a fix that prevents it from crashing when using a polygonal-type infill like Concentric or an even number of multiplied infill lines, together with Randomize Infill Lines.

jellespijker, 2020-06-18:
Looks like https://github.com/Ultimaker/CuraEngine/pull/1206 fixes it

closed 2020-06-21 by jellespijker