Ultimaker/Cura #6703: Cura 4.4 Not Loading

thealphagenius, 2019-11-23:
**Application version**
(The version of the application this issue occurs with.)
4.4

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)
Windows 10 Home 64 Bit

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)
None

**Reproduction steps**
1. Something you did.
2. Something you did next.

I installed Cura.
I attempted to launch Cura.

**Actual results**
(What happens after the above steps have been followed.)
Nothing really happens. I see the Cura logo flash in my taskbar and then it disappears. Then nothing else ever happens.

**Expected results**
(What should happen after the above steps have been followed.)
Cura should load.

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)
The log files are blank. I have attached my DxDiag logs.
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/3882337/DxDiag.txt)

rburema, 2019-11-25:
What are your firewall settings? We use localhost+port to connect to the engine.

We've had a couple of startup issues lately, that mostly seem to be traceable outside of Cura. A user was kind enough to write a step-by-step workaround guide, here: http://embscomputerart.com/fixing-cura-4-3-0/  (this is true for 4.4 as well).

Another thing: Is this a laptop and are you running it with attached screens through a dock?

Otherwise I'm afraid it's the Intel graphics chip. Are the drivers up to date?

thealphagenius, 2019-11-25:
How do I check my firewall settings? I am not sure how. I plan to try the workaround in a bit. This is a laptop, but I am not using attached screens. Also, I just updated all my drivers, including the graphics last week.

 

Jason L. Amerson

ankumara, 2019-11-26:
Tried rbuema solution with firewall. it works for 4.3, and 4.4. kudos!

thealphagenius, 2019-11-26:
I tried that and it is still not working. My drivers are all up to date too. It used to work just fine with this computer and Cura is the only thing giving me issue right now. I am about to just start using different software because this is very annoying.

 

Jason L. Amerson

rburema, 2019-11-26:
@ankumara Good to hear.

@thealphagenius Could you send us the logs? They're in `%appdata%/cura/4.4/cura.log`. If we're lucky, there might be some hints there as to what's going wrong.

thealphagenius, 2019-11-26:
There is nothing there. The folder only has a plugins folder which is also empty. I do have a stderr.log and a stdout.log but both files are empty.

 

Jason L. Amerson

snaiperr, 2019-11-27:
Dear All! The problem related to the COM ports which you have on your PC! Thus, during the Cura initialization the Cura is trying to connect to the printer with the first found COM port, which can belong to the other device(e.g. your BT devices) which are not responding due to disconnection from the PC(laptop). 

Thus, I suggest to remove the auto connect function to the printer(or make it doable manually) as the possible solution.

thealphagenius, 2019-11-27:
That sounds nice and all, but you are going to have to provide instructions on the fix.

 

Jason L. Amerson

Ghostkeeper, 2019-12-04:
You could try going into Program Files where Cura was installed, and under the plugins folder you can delete the USBPrinting folder. This will prevent Cura from checking the COM ports for any 3D printers (and is doable without being able to start Cura).

uwsitsupport, 2019-12-19:
We've found this issue with 4.2.1 and 4.4 on our school machines. We have some machines working and some not despite them being the same make and model. After looking for differences we found the laptops which were not working have a newer Intel driver.

GPU: Intel UHD Graphics 620
Working Driver: 23.20.16.5018 (Released 27/03/2018)

I had one laptop running a 08/2018 driver and one running a 09/19 driver and both failed to load Cura. With the same issue as above, appearing on a task bar briefly and then failing to load. It tends to use up 65MB of RAM before dropping to around 50MB. No log files would be created.

thealphagenius, 2019-12-19:
Not having this issue anymore now that I switched all of my computers to Linux.

Dr. Jason L. Amerson

closed 2024-12-07 by github-actions[bot]