Ultimaker/Cura #6704: Reset post processing features on new prints, please...

Mikerhinos, 2019-11-23:
**Is your feature request related to a problem? Please describe.**
(A clear and concise description of what the problem is. Ex. I'm always frustrated when [...])

**I'm always frustrated when I use Cura for a slice and the last time I used it, it was for a multi colored print so I have M600 "change filament" Gcodes in it.
Then no matter what I want to slice and print next, it may be several days later, it just ruins it while printing because the M600 are still here and it's pretty frustrating :(** 

**Describe the solution you'd like**
(A clear and concise description of what you want to happen. If possible, describe why you think this is a good solution.)

**Post processing reset on each Cura's launch.**

extesy, 2019-11-24:
I disagree. I use post-processing to gradually decrease bed temperature for top layers. I want the same configuration applied on every single print I do.

Mikerhinos, 2019-11-24:
Ok so maybe a "Remember" checkmark on added post processing features, if unchecked, setting isn't saved for next prints ?
M600 pauses the print, nozzle goes (0,0) on bed, its heating stops, then 40cm of filament is ejected by the extruder. Having this happening mi-print because you forgot that last time you used Cura you printed a dual color print for example, is very annoying :(

extesy, 2019-11-24:
> Ok so maybe a "Remember" checkmark on added post processing features, if unchecked, setting isn't saved for next prints ?

Yeah, that would be the best solution to this.

rburema, 2019-11-25:
This seems to me it's a duplicate of https://github.com/Ultimaker/Cura/issues/6597

The conclusion from that thread: If any post-processing script is on, it'll _already_ show a hammer and wrench icon next to the slice-button area.

Mikerhinos, 2019-11-25:
@rburema You're right, it seems to ask for the same thing.
And damn, I never saw that hammer and wrench icon previously...For me the best solution would be a checkmark "Save between prints", or at least a "Are you sure you want to slice ? Post processing features are activated" warning message when clicking on "slice" button :( 
Something that you will see even a sunday morning at 3 AM when you launch a 8h print before going to bed, and when you get up 7h later, the print just stopped on 4th layer because you forgot you made tests on a 10mn print before :(

Goodfeat, 2019-11-26:
I also wanted some scripts to be deleted.
I have several scripts in post processing, some of which are needed for only one print.

can make marked scripts be deleted when the program is closed?

rburema, 2019-11-27:
I've discussed this with my colleagues (again), and the largest problem (in addition to where the scripts are saved right now, which is in the machine) is that post-processing scripts need to be saved to projects.

So, load a model, change settings, put in a 'one-time' post processing script, save to project. What should the behaviour be on opening? Should it just go away afterwards? That reduces the predictability of things saved to projects (and the use of saving to a project in general) since you expect that opening a project, then slicing twice, would give the same results.

closed 2019-11-27 by rburema