Ultimaker/Cura #6705: [4.4-PPA] - Can't scroll in Installed tab of Marketplace

kepbrown, 2019-11-24:
**Application version**
4.4-PPA

**Platform**
Ubuntu 19.04

**Printer**
Ender 3

**Reproduction steps**
Open Cura
Click Marketplace
Click Installed tab

2. Something you did next.
Tried to use the mouse to scroll the window

**Actual results**
Nothing

**Expected results**
The body of the window should allow you to scroll through all the installed plugins

**Additional information**
The Plugins and Materials tabs scroll as expected.

smartavionics, 2019-11-24:
Hello. I can't offer any help with using the PPA but an alternative build based on the master branch (with my additions and bug fixes) can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file there to learn about my builds.

rburema, 2019-11-25:
Also important to note: We support only the AppImage version (when it comes to Linux). Does this bug occur for that as well? If not: for bugs specific to PPA versions of Cura, please contact the maintainer for that package (and close this issue).

Ghostkeeper, 2019-12-04:
Unable to reproduce this problem so far.

closed 2020-06-19 by no-response[bot]