Ultimaker/Cura #6706: [Cura 4.4] Material / Initial Layer Flow not applied to all objects

BoHoehne, 2019-11-24:
The menu entry in material says "Initial Layer Flowrate". 
The user assumes this is the setting for the first layer printed and applies to ALL objects, but it isn't. 
This leads to messed up first layers, especially on weakly sticking printbeds. 

[Reproduce]:
1. Place a box in cura (or anything with a suitable bottom face). 
2. Check that "Top/Bottom flow rate" is set to 100%.
3. Set "Initial Layer Flow" to 300%.
4. Slice it.

The "Bottom layer" extrusion width of the box face will be printed with the "Top/Bottom flow rate" instead of the "Initial Layer Flow rate" setting. This is not suitable for the first layer.
This overrules the settings for the "Initial layer". 

Fix:
Make "First layer flow" setting overrule ALL other flow settings for the first layer of ALL objects.

rburema, 2019-11-27:
I can't reproduce this:

Intial flow 100%:
[image]

Initial flow 300%:
[image]

Could you send a project file (saved, not exported, zip or rename to attach in github)?

BoHoehne, 2019-11-27:
This is a extreme example.
It should not be possible to have such a configuration on the inital layer.

Hope the file helps.

Have a good day.

[TestCube_FirstLayerProblem.zip](https://github.com/Ultimaker/Cura/files/3897457/TestCube_FirstLayerProblem.zip)

[image]

Ghostkeeper, 2019-12-04:
The initial layer flow rate is multiplied with the flow rate of the object. In your case you've set the top/bottom flow rate to 5%. Multiplied by 300% this makes for a total flow rate of 15%. That's what you're seeing in your screenshot.

If you change the top/bottom flow rate to 33% then the total flow rate ends up at 99% which looks approximately normal. You can see the result in this screenshot:
[image]

So far it looks like the behaviour is as expected.

BoHoehne, 2019-12-07:
As a simple user I do not expect the TOP/BOTTOM settings to overrule the first layer settings.
When I dial up the "Initial layer flow" to 300% I expect ALL lines in the first layer to be 300% of the nominal flow rate.

The first layer should be treated as a completely different animal. 
First layer settings should OVERRULE all other settings (for 1st layer), because adhesion has minimum requirements.

As a normal user, I do not expect the settings of the upward layers to influence the initial layer.
The current implementation makes it really difficult to foresee the actual amount of filament that is extruded on the initial layer (if we are talking about the 1st layer).  

I had many failed prints because of such issues.

Ghostkeeper, 2019-12-09:
> As a simple user I do not expect the TOP/BOTTOM settings to overrule the first layer settings.

They don't. Both apply.

> When I dial up the "Initial layer flow" to 300% I expect ALL lines in the first layer to be 300% of the nominal flow rate.

I don't. I'd expect ALL lines in the first layer to be 300% of the *flow rate they'd have when you set it to 100%*.

> As a normal user, I do not expect the settings of the upward layers to influence the initial layer.

Then be warned, that the Wall Line Width affects the first layer as well. The Print Speed affects the first layer too. And even "Enable Support" causes support to be printed on the first layer, because that's necessary to support the layers above it. In general our users expect their settings to be applied everywhere, including on the first layer.

We've chosen for a system that is more adaptable here, rather than a system where it's easier to calculate how exactly Cura arrives at a certain line width. This allows experts to tune the settings better and share their profiles, so that other users can stick to the basic or advanced settings if they find it too difficult to understand.

closed 2024-11-14 by GregValiant