Ultimaker/Cura #6707: CURA 4.4.0 Program freezes  at each action

OUPS65, 2019-11-24:
**Application version**
4.4.0

**Platform**
Spectre, Windows 10 family 64 bits, Intel I7 6500 CPU, 8Go

**Printer**
PRUSA MK2

**Reproduction steps**
1. Something you did.
2. Something you did next.

**Actual results**
[BUG.zip](https://github.com/Ultimaker/Cura/files/3883656/BUG.zip)

He,
I just installed the new release. 
Program freezes (« not responding »)  during  5 à to 10 seconds at each action / profile param value update, insert or delete or move any item on the build plate, record profile, etc. 
Loading a project is so so slow.
De-installed with Revo Uninstaller and re-installed : no change.
I do not have any of these problems with my current 4.3.0  version fortunately still active.
Sorry
Philippe

smartavionics, 2019-11-24:
Hello @OUPS65 , I do not know for sure what the problem is but I am guessing that it could be something that I have already fixed in my Cura builds. My builds can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0 and they may be installed alongside the UM releases without conflict. Please read the README.md file there if you wish to learn more.

Ellecross, 2019-11-25:
crosspost from the [forum](https://community.ultimaker.com/topic/30191-ultimaker-cura-44-released-with-new-intent-profiles/?do=findComment&comment=251308). :D FYI

OUPS65, 2019-11-25:
Correct. I’m not sure where to call for help when a new program version is involved.
solution from the forum :
Tks, @RoboDLC
What I made, based on your suggestion:
norton -> pare feu -> controle programme -> ajouter ( add a new programme autorisation) -> selected CURA 4.4.0 and gave full autorisation.
Now, the problem is solved.

rburema, 2019-11-27:
@OUPS65 Usually (read: we prefer) here, but in this case there's more information/an already ongoing discussion on the forum.

@Ellecross Don't forget to attach the ticket number to here as well if we make one! :-)

closed 2024-12-07 by github-actions[bot]