Ultimaker/Cura #6708: Where are the visible options stored?

FlailAway, 2019-11-24:
Mint 19.3+Cinnamon

Further to my now "deferred," questions of a few days back, https://github.com/Ultimaker/Cura/issues/6696 I have some questions for an alternative approach.

Where is the status of all those check boxes stored? I have searched ".local" and ".config."

I cannot find anything that has information to set those boxes the next time I run the Appimage of Cura 4.4.

I'd be happy to write a GUI-program (Public Domain) to manipulate those check boxes manually and save the results under user-defined names that can be renamed to the Cura-default and loaded at the next Cura start. Gotta be easier than spending 15-minutes manually resetting only the items the user wants to see.

Give me a text file of that full list of options, plus where they are stored and in a day or so, as my Aussie Mate says, "Bob's yer Uncle."

fieldOfView, 2019-11-24:
> I'd be happy to write a GUI-program (Public Domain) to manipulate those check boxes manually and save the results under user-defined names that can be renamed to the Cura-default and loaded at the next Cura start. 

There's a plugin in the Marketplace that will let you save custom "setting visibility sets". You might find that useful too.

The visible settings list is stored in cura.cfg. I forgot where that is on Linuxes, but the folder should open with Help -> Show configuration folder.

smartavionics, 2019-11-24:
I think you need to install this...

[image]

FlailAway, 2019-11-24:
Thanks guys, 'tis a pity no one mentioned that the other day with my first gripe. :)
I had searched all of the likely Folders before posting this morning and while I did find a cura.cfg in .local or .config, but it only had a small amount of plain text, maybe 100-Bytes.

closed 2019-11-24 by FlailAway

fieldOfView, 2019-11-24:
It is a matter of asking the right question, I guess.