Ultimaker/Cura #6709: [FQ] Region printing

yet-another-average-joe, 2019-11-25:
**Is your feature request related to a problem? Please describe.**

For complex models, region printing is convenient. But the model has to be cut with the creation software (CAD, blender, or whatever). But doing this, custom supports or support blockers have to be recreated each and every time, and when the full model will be printed. 

**Describe the solution you'd like**

Region printing for the sliced model would be usefull, keeping all custom parameters : regions, for custom supports and support blockers, scaling, etc.

**Describe alternatives you've considered**

The only alternative in Cura is lowering the model and the custom/blocking regions, and then stopping the printer when the needed height is printed. Example : a head, and we need to experiment with the nose : doing this, a whole useless head section will be printed with the nose. A boatload of plastic could be saved...

**Affected users and/or printers**

All users and printers

**Additional context**

It could work the way custom supports and support blockers work : a region cube. Cura could do a boolean operation, excluding everything outside. The UI is already in Cura (region functions). It would need a button below "Custom supports" : "Select region". And some code (I am unable to write !). I have no idea of the complexity for boolean operations on STL files...

nallath, 2019-11-25:
You can already do this. Add a cube (or any other model that spans the area you don't want printed). Select the per object settings. Select the "Modify settings for overlaps". Ensure "infill only" is not selected.
Change the settings so that nothing is printed (eg; wallthickness to 0, top bototm thickness to 0, infill to 0%)

yet-another-average-joe, 2019-11-25:
Oh yes ! Didn't know this trick.

(at first I didn't understand the answer, as you wrote "area you don't want printed", and it's about area to be printed : voxels that belong to volume A and volume B :  A ∩ B)

I just discovered this feature following your instructions ! But there is a tiny issue : build plate adhesion options are not available : for example, a brim will be created for the two models (A U B), printed or not : we cannot get a brim for A ∩ B only. Ticking "Show all" in the dialog shows the "Build plate adhesion" menu, but it is empty.

I was thinking of something different : there is still a volume to be defined outside of Cura, while Cura already has a tool that creates parallelepiped volumes. Cura could convert a "Custom support" or a "Support blocker" to a real volume, do the "A ∩ B" operation, and automatically set "A XOR B volume" to zero wall, infill, and top/bottom thickness.

So : Cura has a volume creation + editing tool + boolean functions. The only things missing are a conversion to real volume and 3 auto settings for the excluded regions.

[image]

Ghostkeeper, 2019-12-05:
The modifier meshes are meant to modify settings in part of your print. They are not meant to cut away pieces of your meshes. This falls into the realm of CAD software which is out of scope for Cura. We will not implement this.

closed 2019-12-05 by Ghostkeeper