Ultimaker/Cura #6710: set parameter problem

lyfbuaa, 2019-11-25:
Application version
Ultimaker_Cura-4.3.0.AppImage

Platform
fedora 31 x86_64

Printer
diy printer like ultimaker with marlin 1.0.2-2

Reproduction steps
1. shell-》bottom thickness : 24 mm
maybe tell me how to set <240 layer  solid

Actual results
<240 layer  hollow

Expected results
<240 layer  solid

[image]

nallath, 2019-11-25:
I've tried to reproduce your issue, but when I set the bottom thickness to 24 I get a lot of very thick layers. 
Could you share your project file? It could be that you have other settings changed / set that are interfering with this.

lyfbuaa, 2019-11-25:
en,sorry.  please close it,I find it's the prolem of my cura .

closed 2019-11-25 by nallath