Ultimaker/Cura #6711: Add favorite options please (star mark)

ZhenyaKa, 2019-11-25:
Hiding some options in list is good, but in 99% cases we need to change 2-5 options.
Now I need to scroll list to find it or use very slow filter.

If I'll have any "star-button" near each line of options, what move that option to top of the list I will be happy!

nallath, 2019-11-25:
You mean like the custom visibility plugin?

ZhenyaKa, 2019-11-25:
> You mean like the custom visibility plugin?

I can't find that plugin.

I need 5-10 marked options at top of global list, which I can mark any method from main list.

For fast changing them and not scrolling each time

rburema, 2019-11-27:
I think @nallath meant the 'Setting Visibility Set Creator', available in the Marketplace (the 1st button in the top right, just above the settings fold-out).

Ghostkeeper, 2019-12-05:
I think the proposed solution (or workaround if you see it that way) works fine for this use case.

closed 2019-12-05 by Ghostkeeper