Ultimaker/Cura #6712: Add flag that group title can be filtered

ZhenyaKa, 2019-11-25:
If I need to go to Support group for example, I need to scroll to that group.

I can write "support" in filter field, but that method hide some lines, that not include that word.

It will be good if will be posibility to filter group header without hiding lines that this group include.

Hot keys like ctrl+1, ctr+2... etc for fast moving to each group will be good solution too (with key combination hint).

Ellecross, 2019-11-26:
Hey @ZhenyaKa thanks for the comment, I'll ask the team ;)

Ghostkeeper, 2019-12-05:
@Ellecross What was the result of asking the team?

ninovanhooff, 2019-12-30:
I see this request was referenced in an internal ticket CURA-6959: "Collapse all" for all settings categories.

So we looked at @ZhenyaKa problem which is: "Getting to settings category X takes too long" and we are addressing this by having a way to collapse all categories.

evtrados, 2020-01-03:
Closing this issue, since the "Collapse All Categories" option is addressing the original problem.
This option will be available in the next release.

closed 2020-01-03 by evtrados