Ultimaker/Cura #6713: Clearing option filter frozes cura

ZhenyaKa, 2019-11-25:
I write in filter field "support" for example.

When I clear that word I got freeze while list building - about 5-7 seconds.

Ellecross, 2019-11-26:
Which version are you using? @ZhenyaKa

ZhenyaKa, 2019-11-26:
> Which version are you using? @ZhenyaKa

this freeze appear in all versions including 4.4 beta

fieldOfView, 2019-11-26:
How many settings do you have visible?
Does the same freeze also happen if you have the "Basic" set visible (from the hamburger menu next to the filter box)?

ZhenyaKa, 2019-11-26:
I have almost all settings visible

rburema, 2019-11-27:
> ... including 4.4 beta

Also in 4.4.0 (final) itself?

fieldOfView, 2019-11-27:
> Does the same freeze also happen if you have the "Basic" set visible

Ghostkeeper, 2019-12-05:
Seems to me that the problem is to create 495 text fields, labels, mouse areas, reset buttons, function restore buttons, setting value labels and unit labels.

closed 2024-12-07 by github-actions[bot]