Ultimaker/Cura #6714: Materials list doesn't update when a material is created or deleted

acsdog, 2019-11-26:
**Application version**
4.4.0

**Platform**
OS Name                           Microsoft Windows 10 Home	
Version	                          10.0.18362 Build 18362	
CPU                                    Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz, 3696 Mhz, 6 
                                           Core(s), 12 
                                           Logical Processor(s) 
GPU                                    NVIDIA GeForce GTX 1070 Ti
Graphics Driver Version     441.20

**Printer**
Monoprice MP Select Mini V1

**Reproduction steps**
1. Click Settings > Extruder 1 > Material  > Manage Materials
2. Click "Create"

**Actual results**
New material doesn't appear until the window refreshes (e.g., opening the "Printers" window, then going back to "Materials"

**Expected results**
New custom material should appear in list of materials immediately upon clicking "Create".

**Additional information**
Occurs whenever a material is created or deleted. Method of navigating to "Materials" window has no effect on this bug.

Ellecross, 2019-11-26:
Thank you for reporting this! 

It is indeed a bug. We'll discuss this with the team.

evtrados, 2019-12-13:
The bug has been fixed and in version 4.5 of Cura it will not exist anymore.
As a result, I am closing this issue. Thank you again for reporting it @acsdog!

closed 2019-12-13 by evtrados