Ultimaker/Cura #6715: post proces script linked to quality definition

maukcc, 2019-11-26:
Is it possible to incorporate a post process script to quality definition files?
example:
If we choose a specific quality setting, it will automatically add the "Time Lapse" script with values.

nallath, 2019-11-26:
Currently there is no way to do this. The profile system knows nothing about the post processing.

I also don't see a quick way to incorporate that feature into the profile system.

Could you perhaps explain the complete scenario why you would want this? I might be able to provide a different solution.

maukcc, 2019-11-26:
The way we came to this is that we wanted to do a Nozzle wipe (G12) every xx seconds or layers

nallath, 2019-11-26:
I'd say that's something the engine should do (and not a post processing script).

rburema, 2019-11-27:
There is something similar (but implemented without the `G12`, I think) in the experimental settings: Turn on 'Wipe Nozzle Between Layers' and a number of related options should appear.

maukcc, 2019-11-27:
@rburema We did not see that one, thanks. unfortunately that is not usable for us. We will make a new issue about this.

closed 2019-11-27 by maukcc