Ultimaker/Cura #6716: Under Extrusion Cura 4.4 Ultimaker 2 and more

MicheliDE, 2019-11-26:
Hi All,

I have been having some different issues with cura 4.4 on my ultimaker 2. I have not upgraded it to the 2+. With Stock Cura 4.4 and Firmware uploaded to the printer from Cura 4.4 I am getting under extrusion, line width problems, printing every other line for bottom and top layers. This is using stock profile and no support checked.

I have selected ultimaker 2 as my print machine but It seems like it thinks I have a 2+ and not a 2 with the amount of under extrusion I am getting.

I will likely go all the way back to a very early version of cura and firmware until this issue is resolved. Does anyone have the last cura and firmware that I can use that works best with ultimaker 2?

rburema, 2019-11-27:
This came up in a chat with @nallath yesterday. A very likely cause of this on UM2 (not +) is damage to the to the PTFE Coupler. There was an issue that crops up that causes the coupler to be stretched out over months of print. The 'cure' for this is to replace the coupler with a new one.

> I will likely go all the way back to a very early version of cura and firmware until this issue is resolved.

If that's the case, going back to older versions or changing the firmware is unlikely to help.

MicheliDE, 2019-11-27:
Hi Rburema,

I think i found the issue and am making a test print now. I had some material that was labeled as PLA from the manufacturer when in fact it was ABS! This made the material not spread out as much because it wasn't extruding at the right temperature. I use ColorFab and Formfutura for material and in some cases I have found that spools are accidentally mixed! So the machine is printing along with PLA and the next part of the spool is ABS and so it starts under extruding and some times clogs up the tip mid print! I have also had the reverse where it is ABS and the material starts burning mid print and clogs the tip. Looks like this print is going well on cura 4.4 now that I have tweaked the temperature.

rburema, 2019-11-27:
Hi @MicheliDE 

Good to hear! (Well, for us at least ... mislabelling is a serious issue!)

I'll close this now. If your test-print fails, or for whatever other reason you change your mind, feel free to re-open.

closed 2019-11-27 by rburema