Ultimaker/Cura #6720: Nozzle wipe feature

maukcc, 2019-11-27:
The nozzle wipe feature relies on a X position to be set by the user, and starts on the last Y position of the program. 
We would like to set the Y position also.

Even better would be a "nozzle wipe gcode" to be set with machine settings.

Ellecross, 2019-11-27:
I'll bring this up to the Cura team

nallath, 2019-12-02:
For our reference; CURA-7015

fvrmr, 2021-06-04:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-07-13 by github-actions[bot]