Ultimaker/Cura #6721: [4.4.0] Can't open an OBJ file

eranlahav, 2019-11-27:
[obj.zip](https://github.com/Ultimaker/Cura/files/3896076/obj.zip)

 
**Application version**
4.4.0

**Platform**
  Model Name:	MacBook Pro
  Model Identifier:	MacBookPro13,3
  Processor Name:	Quad-Core Intel Core i7
  Processor Speed:	2.9 GHz
  Number of Processors:	1
  Total Number of Cores:	4
  L2 Cache (per Core):	256 KB
  L3 Cache:	8 MB
  Hyper-Threading Technology:	Enabled
  Memory:	16 GB

**Printer**
Ultimaker 2 ext+

**Reproduction steps**
1. just tried to upload the file

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

smartavionics, 2019-11-27:
The files contain some lines that have a \ in them and Cura doesn't like that.
```
vn 5.533322137277522e-17 -0.9967947006225586 0.08000191301107407
vn 0 -0.9967947006225586 0.08000191301107407
vn 1.106664427455504e-16 -0.9967947006225586 0.08000191301107407
vn -2.213328854911009e-16 -0.9967947006225586 \
 0.08000191301107407
vn -2.213328854911009e-16 -0.9967947006225586 \
 0.08000191301107407
vn 2.213328854911009e-16 -0.9967947006225586 0.08000191301107407
vn 2.213328854911009e-16 -0.9967947006225586 0.08000191301107407
```

smartavionics, 2019-11-27:
That's a Cura bug as it should cope with those \ line continuation characters.

smartavionics, 2019-11-27:
I have produced a simple fix that let me read the files...

[image]

I will submit the fix as a PR so that the Cura devs can use it. I'm not a Python programmer so they may decide to do something else but at least it identifies the problem.

rburema, 2019-11-28:
Normally it takes us a while to review PR's, but this one was small (and more importantly, low risk) enough that I could do it on my own (sometimes it's better to ask forgiveness than permission after all).

closed 2019-11-28 by rburema

eranlahav, 2019-12-01:
Green question: how can I know this fix is deployed to Cura?

smartavionics, 2019-12-01:
As it has been merged into the master branch it should appear in the next major release (4.5?)

eranlahav, 2019-12-01:
does it have Release to Customer date ? or a roadmap timeline ?

smartavionics, 2019-12-01:
No, but if you are eager to try this you can install one of my Cura builds that has the fix already. You can find my builds at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. My builds can be installed alongside the UM releases. Please read the README.md that is there.

fieldOfView, 2019-12-01:
Ultimaker Cura releases are normally two months apart. So I would guesstimate end of January for the next release.