Ultimaker/Cura #6723: Print Nozzle "Strumming" Support When Moving

coofercat, 2019-11-27:
**Application version**
Cura 4.2.1

**Platform**
Mac Mojave

**Printer**
Ultimaker Original

**Reproduction steps**
1. I downloaded the STL for the roll rings from here: https://www.youmagine.com/designs/rollrings
2. I selected the default "fine" printing settings, and clicked "Generate Support"
3. Printed the part (with PLA, on a heated bed, single head, only one material)

**Actual results**
The print failed about a third the way through because a bit of the support inside one of the rings had been pulled upwards by the head. This then obstructed the fan duct, which came loose, and pulled up the support further. Once all that had happened, the support had moved so couldn't be printed on again.

I've observed that the head seems to "strum" the zigzag of the support structure when moving from one side of it to the other. It seems support is possibly slightly higher than the main part when printing.

Less easily reproduced, but I also found issues when printing the Team Unlimbited arm (the palm of the hand, especially) - find it here: https://www.teamunlimbited.org/arms. This print was successful, but the head seemed to cross the support about the same place on each layer, which made a nice "bbbrrriing" noise as the head strummed the lengths of support, and also formed a sort of "furrow" in the support as it did so. Over many layers, there was an obvious defect in the support in that area (in this case, the support didn't detach and the print was successful. 

**Expected results**
(obviously, I'd have liked the print to succeed)

I'm wondering if there's a way to (perhaps optionally) have the head either avoid the support altogether, or else find a way to navigate it without touching it.

Alternatively, could support be printed after the rest of the layer? It seems that printing it first means it forms an obstruction to moving around to reach the rest of the print, so maybe do it last instead?

I also wonder if this also has something to do with the "angle of incidence" between the head and the shape of the support. In the Roll Rings, in one area the support is diagonal with respect to the head movement, whereas in the other area, it's perpendicular. Presumably hitting it "broadside" has a greater chance of pulling it up than hitting it at an angle...? Could the head move less directly to ensure it doesn't hit the support perpendicularly?

Another option, which I'm loathed to suggest is that where support is narrow (and perhaps tall) such as in the Roll Rings, it obviously doesn't have much adhesion to the model or bed (compared to a wide, flat structure). Perhaps in such cases it needs additional adhesion?

**Additional Information**

Here's a screenshot of the preview of the print at about the point mine failed. Here we can see the printer has printed support, and has started on the three "pilars" of the roll rings. It's done two of them, and is now about to move to the right to do the last of them. To do so, it will cross the (newly printed) support, and at some point will (in my case) dislodge the top left most end of it enough that it protruded upwards and obstructed the fan shroud.

[image]

rburema, 2019-11-28:
I see what you mean. I'll discuss it with the team.

In the mean-time, I think I found a workaround: (Assuming 'Avoid printed parts when travelling' and 'Avoid supports when travelling' are already on). Make the 'Travel avoid distance' higher, so from 3 mm to 4 mm for example.

coofercat, 2019-11-28:
oh! I looked for an "avoid support" option, but didn't see one. I've checked it now, and the simulation shows the head running slightly to the side of the support structure rather than over the top of it - so that may indeed be a good workaround (incidentally, my default "travel avoid distance" is 0.625mm - I'll try it with that before changing it). I'm running a print now, so we'll see in a few hours...

rburema, 2019-11-28:
Good to hear! Please close the issue if your problem is resolved.

coofercat, 2019-11-28:
Good news: With "Avoid supports when travelling" (set to 0.625mm), the print has been successful.

There's some unnecessary retracting going on, which makes the support a lot less solid than it probably should be in one particular place - but it works.

As for this bug - I'm happy to close with "User is being pedantic", but I think there's still something to be done here. Some options:

1) Select "Avoid support when moving" by default
or
2) Move the "Avoid support when moving" from the "Travel" section and put it into the "Support" section, optionally select it by default when "Generate support" is ticked
or
3) Look at the way the slicer creates support and see if it can be done at the end of the layer or in some other way so it does not form an obstruction to moves

rburema, 2019-11-28:
> 1. Select "Avoid support when moving" by default

This (sort of stuff) is handled in the printer profiles/definitions. We're happy to receive pull requests for 3rd party printers!

In fact, the S-line Ultimakers (S5 and the new S3) and the 'new' Creality profiles already have 'travel_avoid_supports' forced to true. Otherwise it is on if and only if either the user manually sets it _or_ ['retraction_combing' is not set to 'off' _and_ 'travel_avoid_other_parts' is true]. The default values of which depend on which printer (definition) is used.

> 2. Move the "Avoid ...

It's a child option of another setting that doesn't have much to do with support. I think this would be confusing.

> 3. ... see if it can be done at the end of the layer ...

It'd still bump into _other_ support if this fails (which can also cause the print to fail). This would be masking the problem IMHO.

closed 2019-11-28 by rburema