Ultimaker/Cura #6726: [4.4.0] Profile type in profile list have incorrect color

NeoProgAIM, 2019-11-28:
Environment:
- Windows 10 (x64)
- Tevo Tarantula
- Cura Ultimaker 4.4.0 with dark theme

Profile type in profile list have incorrect color. You can see it on screenshots. 
[image]

I think "Custom profiles" section name should be have color equal "Default" section name (gray). May be also add some separator after section name?

Ellecross, 2019-12-02:
@NeoProgAIM hey, thanks for reporting this. @Yian-Lai and I will take a look. There are actually a bunch more that are incorrect in the dark theme that we have to correct. :)

Ellecross, 2019-12-02:
UM Cura devs, we're tracking all dark theme issues in CURA-6941

closed 2024-12-07 by github-actions[bot]