Ultimaker/Cura #6727: Add ability to create two layer support; the 1st with model material, the 2nd with dissolvable material

mbmast, 2019-11-28:
The problem:

Depending on the model to be printed, when Cura generated support is desired, the Cura generated support structure can consume a lot of support material.  This can be expensive if the support is made from any number of dissolvable materials, as dissolvable material can cost several times the cost of the material used to print the model.

The solution:

Dissolvable material is useful (and often necessary) for generating support where removing the support material after a print is completed is either difficult or impossible if the support had been generated using the model material (e.g. PLA, ABS, etc.).  The reality is, in many cases, only a minimal interface of dissolvable material is necessary between the model and any non-dissolvable material used to support the dissolvable material.

The change to Cura I'd like to see:

When printing Cura generated support with a dedicated extruder loaded with dissolvable material, 
I'd like see options in Cura that tells Cura to use the extruder loaded with model material for some percentage of the support height (say 85%) and to use the extruder loaded with dissolvable material to complete the upper layers of the support.  This can save a considerable amount of the more expensive dissolvable support material.

Alternatives to this idea:

Without this added feature, support structures must be made part of the model.  These added support structures would leave a gap between the top of the support structure and the bottom of the model.  Cura would then add the "missing" support using the more expensive dissolvable material.

Affected users:

This will affect those users that have multiple extruders and print larger objects that require large amounts of dissolvable materials used for support.

fieldOfView, 2019-11-28:
Do you know about the “support interface” feature? You can print the bottom and roof of the supports (which touch the model) with dissolvable material and print the rest with the normal material.

mbmast, 2019-11-28:
@fieldOfView I do now!  Thanks.

closed 2019-11-28 by mbmast