Ultimaker/Cura #6728: Infill sometimes goes through walls

MarcusAdams, 2019-11-28:
**Application version**
4.4.0

**Platform**
Windows

**Printer**
Prusa MK2

**Reproduction steps**
Load "Filtr 154 v6-oprava.stl" file. Slice with 15% grid or gyroid.

**Actual results**
Infill appears within outer bounds of model but outside of model walls (within cylindrical holes in model).

**Expected results**
Infill should not appear outside of model walls.

**Additional information**

[Filtr 154 v6-oprava.stl.zip](https://github.com/Ultimaker/Cura/files/3902684/Filtr.154.v6-oprava.stl.zip)
[curaproject.3mf.zip](https://github.com/Ultimaker/Cura/files/3902687/curaproject.3mf.zip)
[image]

smartavionics, 2019-11-28:
Hi, I think I have already submitted a fix for this (https://github.com/Ultimaker/CuraEngine/pull/1156), looks like the same issue.

MarcusAdams, 2019-11-29:
> 
> 
> Hi, I think I have already submitted a fix for this ([Ultimaker/CuraEngine#1156](https://github.com/Ultimaker/CuraEngine/pull/1156)), looks like the same issue.

@smartavionics , it doesn't look like the same issue, but I suppose you could try your fix on the attached STL and see.

smartavionics, 2019-11-29:
OK, here's the result...

[image]

smartavionics, 2019-11-29:
BTW, the above mentioned fix is already in my cura builds which can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. They can be installed alongside the UM releases. Please read the README.md file there if you want more info.

Sharplayers, 2019-12-03:
> BTW, the above mentioned fix is already in my cura builds which can be found at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. They can be installed alongside the UM releases. Please read the README.md file there if you want more info.

thx, your buid solved my problem

archer183, 2020-10-11:
Still an issue in version 4.5 (unclear if it should have been fixed in this version or not)

nallath, 2020-10-12:
The fix that was mentioned is already merged. I'm not 100% sure in what version though, but the latest should have it.

Ghostkeeper, 2020-10-12:
The [relevant commit](https://github.com/Ultimaker/CuraEngine/pull/1156/commits/377ee8f379a986fb81f5226ce9181b175a0492c3) should be in since 4.5. So if it's still an issue in 4.5 that commit didn't fix the problem. Do you have any reproduction steps?

archer183, 2020-10-12:
It is still an issue.   I will upload a saved file showing it tonight.

archer183, 2020-10-13:
[
[infill_crossing_walls_layer138.zip](https://github.com/Ultimaker/Cura/files/5369023/infill_crossing_walls_layer138.zip)

See attached at layer 138.

Ghostkeeper, 2020-10-13:
Interesting model!
Indeed, this bug or something like it reproduces with that model even in current Master. There are two layers where the infill protrudes through the walls:
[image]

The bug seems to only occur with Gradual Infill enabled, so perhaps something weird is still happening there that is not happening with the normal infill area (or something weird is not happening with gradual that is happening with the normal area).

archer183, 2020-10-13:
I can get around it usually by tweaking line width up or down by .001mm on
a per model basis, but what works for one design doesn't necessarily work
for others.

printingotb, 2020-10-27:
I get this in 4.8 beta when I enable "infill support"
[image]
[infill support.zip](https://github.com/Ultimaker/Cura/files/5444711/infill.support.zip)

printingotb, 2020-10-27:
Upon closer inspection, the issue disappears in some areas in my project with a max res value below  0.3mm.
Some areas still have the issue.
Max dev was at 0.05 
[image]

konskarm, 2020-11-13:
This is now in our backlog. Devs see internal ticket CURA-7845.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

archer183, 2024-11-22:
Haven't seen it in a while, so no

GregValiant, 2024-11-22:
A report on the UM Forum just came in regarding this behavior in 5.9.0.
[UM Forum](https://community.ultimaker.com/topic/46385-infill-going-through-wall-in-cura-59/#comment-338933)

Since 5.9.0 is essentially a different app I'll go ahead and close this report as it isn't relevant to the newer versions of Cura.