Ultimaker/Cura #6729: Feature request

lwbsbde, 2019-11-28:
Hi
I have a few suggestions
1. Adjustable Fan speed for overhanging , outer Wall and small Feature
2. Display the fan speed in the preview ( similar to the feed rate in the preview)
3. A button to synchronize the settings from Extruder 1 to Extruder 2 .
4. Dynamic width of the inner Wall Line to avoid gaps in narrow areas.

Ellecross, 2019-12-02:
Can we ask you to fill up the template? This would give us a better background on why you want these features. :D

lwbsbde, 2019-12-02:
1.If the fan speed is too high, the inter-layer adhesion will be reduced.  That's why I would prefer to preset it a little lower, and only increase it on overhangs.  Similar to bridges.

2.To check the setting before printing.

3.To make it easier to adjust the settings for two-color printing on both extruders.

4.I already mentioned.

Ghostkeeper, 2019-12-05:
1. This is a duplicate of https://github.com/Ultimaker/Cura/issues/5140. So I'll close this as a duplicate. Please create separate issues for separate features and fill in the template.
2. This can be done, but of course the fan speed can only change per layer per extruder right now.
3. This already exists. Right click on the settings list to synchronise between the extruders.
4. This is a research topic that we have in progress, but it's expected to take in the order of months or years to complete. It's a complex problem that is also severely limited by the intricacies of the printing process and delay in changes to flow.

closed 2019-12-05 by Ghostkeeper

lwbsbde, 2019-12-05:
Vielen Dank