Ultimaker/Cura #6730: Time Lapse Cura 4.4

lwbsbde, 2019-11-28:
Hallo.

Cura makes an error writing the Time lapse gcode.

Wenn ich "Park print head " aktiviere, muß ich zusätzlich den Retract bei Schichtwechsel aktivieren, um Stringing zu vermeiden.
Leider fährt der Druckkopf nach dem Retract zu der Anfangsposition der Nächsten Schicht. Dann erst zu der Parkposition um das Bild zu machen, und macht hier den Filament  Rückschub ohne vorher auf den Startpunkt zurückzufahren.

01  G1 F480 X115.779 Y160.581 E423.37746                 Letzte Position fertige Schicht
02  G1 F2400 E420.37746                                               Retract                       
03  ;MESH:NONMESH
04  G0 F600 X115.779 Y160.581 Z0.5                            Lift
05  G0 F3000 X128.76 Y150.762                                    Startposition der neuen Schicht
06  ;TIME_ELAPSED:733.128714
07  ;TimeLapse Begin
08  G1 F9000 X200 Y200 ;Park print head                     Parkposition für Snapshot
09  M400 ;Wait for moves to finish                               
10  @make_snapschot xxx ;Snap Photo                        
11  G4 P700 ;Wait for camera                                       Pause
12  ;TimeLapse End
13  ;LAYER:2
14  ;TYPE:WALL-INNER
15  G1 F1500 E423.41904                                              Filament zurückschieben
16  G1 F372 X130.038 Y151.413 E423.47151                 Druck neue Schicht
17  G1 X131.494 Y151.654 E423.52551

Die Zeile 5 müsste nach Zeile 11 stehen.

Ellecross, 2019-12-02:
Hey @lwbsbde back when we merged this PR from @wporter82. @Ghostkeeper already mentioned adding the retract in. See https://github.com/Ultimaker/Cura/pull/4353

I'll check with Cura team what we can do in this case. 

In the future, do you mind filling in the bug template and in English, please? :) Thanks.

Ellecross, 2019-12-03:
Hello @lwbsbde unfortunately, we will have to defer this bug request for now. :( We have a lot of other high priority bugs at the moment and we cannot fix this. However, pull requests for fixes are always welcome! Sorry!

Ghostkeeper, 2019-12-05:
@Ellecross please keep deferred issues open so that they can still be found.

yucelll, 2020-01-23:
Hello i have one problem for this post!
İf i use time lapse processor, some parts missing extrude.
Because running;
Rectract**>**go to for push button**>**wait**>**EXTRUDE**>**go to last position.
I think it should be;
Rectract**>**go to for push button**>**wait**>**go to last position**>** EXTRUDE

Looks:
[image]

mahtDFR, 2020-01-24:
Thanks for flagging. I've reproduced it, the orange line shows an extrusion where it shouldn't be. It's being looked into.

[image]

yucelll, 2020-02-03:
I need this function! And i waiting.

mahtDFR, 2020-02-03:
> I need this function! And i waiting.

Just committed it so it should make it into 4.5 beta (which goes out next Tuesday!)

yucelll, 2020-02-04:
> 
> 
> > I need this function! And i waiting.
> 
> Just committed it so it should make it into 4.5 beta (which goes out next Tuesday!)

This is perfect! thanks.

closed 2020-02-04 by mahtDFR