Ultimaker/Cura #6731: Cura 4,4 some custom materials cannot be removed

V3DPrinting, 2019-11-29:
**Application version**
(The version of the application this issue occurs with.)
Cura 4.4.0

**Platform**
(Information about the operating system the issue occurs on. Include at least the operating system. In the case of visual glitches/issues, also include information about your graphics drivers and GPU.)
macOS 10.14.6 on iMac 2017

**Printer**
(Which printer was selected in Cura? If possible, please attach project file as .curaproject.3mf.zip.)

UM S5 or UM3E

**Reproduction steps**
1. Something you did.

Migrated materials from 4.3 to 4.4 by upgrading the software.

2. Something you did next.

Trying to remove a material

**Actual results**
(What happens after the above steps have been followed.)

The "Remove" button is not usable and the material cannot be deleted 

[image]
**Expected results**
(What should happen after the above steps have been followed.)

Material should be deletable 

**Additional information**
(Extra information relevant to the issue, like screenshots. Don't forget to attach the log files with this issue report.)

If you create from the generic material a custom material in 4.4 it is impossible to remove it
Same if you duplicate a custom material

V3DPrinting, 2019-11-29:
I forgot to state that the material is not activated when trying to remove it

V3DPrinting, 2019-11-29:
And the configuration folder
https://www.dropbox.com/sh/y0gdmw76onfw8h4/AACLe6B-PLjHSUYYizYOPWRUa?dl=0

nallath, 2019-12-02:
If you have a material active anywhere, it can't be removed.

Ellecross, 2019-12-02:
Indeed, this is expected. Can you check if the material is selected in one of your machines, please @V3DPrinting ?

V3DPrinting, 2019-12-02:
I do confirm there is a bug against the expected result.

I have only one printer loaded and running at the moment

[image]

If I try to remove a material which is not in use at the moment, it is not possible for some of them (created or modified with Cura 4.4)

[image]
[image]

On the other hand, a custom material, based on Generic PVA, which is currently loaded in the printer can be removed.

[image]

Now, this feature is a non sense, as we need to create all our custom materials based on Generic material profiles, we won't be able to remove ANY material if the type is already loaded in one printer.

If you got a farm of printer (I'm about to have 5 ultimakers) there is nearly no chance to have the type of material not in use.

Please remove this feature ASAP!

nallath, 2019-12-02:
I can't reproduce this, sorry. The only way to do this is when I have a machine in Cura (active or not) that has the material selected.
Did you remove all other machines before testing this? 

As for your example, your printer has FormFutura Helios support active, not FormFutura Aqua solve (which is also why you can remove it)

V3DPrinting, 2019-12-02:
Hi Nallath

My current machine configuration

[image]

So I should be able to remove a Nylon, which is not the case. Either PA CF or PA CF 2 which are based on generic Nylon.
Currently it is not possible

[image]
[image]

If I understand well your feature, it's based on the material type, so I shouldn't be able to remove any PVA or custom material based on Generic PVA. Correct ?

Currently I would be able to remove the Aquasolve which is based on Generic PVA, which is currently loaded in the printer.

[image]

It is normal not to be able to remove a material which is currently active in Cura.
But we should be able to remove from Cura a material which is currently loaded in a printer, as the printer uses the Generic materials.

Again your feature is not properly implemented and not valid in terms of user experience.

nallath, 2019-12-02:
> So I should be able to remove a Nylon, which is not the case. Either PA CF or PA CF 2 which are based on generic Nylon.
> Currently it is not possible

I don't think you understand what I'm trying to say. You can't remove a material if it's active on a machine that you _locally_ have within Cura. What setup Connect has, has little to do with this.
 
> If I understand well your feature, it's based on the material type, so I shouldn't be able to remove any PVA or custom material based on Generic PVA. Correct ?

You misunderstood. It's based on the exact material match. So if you have two machines in Cura, the active one has 2x "your newly created PLA" and the inactive one has 2x "My older self created PLA", I will be unable to remove the newly created and the older PLA, because there is still a configuration using it. 

> It is normal not to be able to remove a material which is currently active in Cura.
> But we should be able to remove from Cura a material which is currently loaded in a printer, as the printer uses the Generic materials.

Inactive printers (eg; A printer that is in the list, but not the active one) also influence what can and can not be loaded. What the machine actually has loaded doesn't matter.

V3DPrinting, 2019-12-02:
Nallath

Thanks for the clarification.

Based on what you said, in my current configuration, I don't have any PA CF, Nylon type, loaded in any printer nor active in Cura.
So I should be able to delete it, currently it's not the case

So there is a bug

nallath, 2019-12-02:
Right, that was what i was trying to figure out. What you found is indeed a bug. I've already added a one thing that should fix a part of the issue. I'm still looking into the rest.

V3DPrinting, 2019-12-02:
Great !
Unfortunately that doesn't make my day....
I will revert back to 4.2.1 today after checking another weird behavior

Ghostkeeper, 2019-12-05:
The bug here seems to be that new printer stacks are created behind the scenes upon switching printer type in a cluster of printers, and those are never deleted and keep the material in use. This not only prevents deleting those materials but also makes Cura slow down a lot.

Ghostkeeper, 2020-05-07:
These hidden stacks are properly deleted again in Cura 4.5. This fixes the bug. However make sure that if you have a cluster in Cura with more than one different type of printer, the material is not loaded for any of them (using the drop-down just above the nozzle/material selection). If it is, the material can't be deleted.

V3DPrinting, 2020-05-07:
@Ghostkeeper I do not agree with the closing.
The bug is still there in 4.6.1.
My cluster is 1 UMS5 (host) and 2 UM3E.
Materials loaded in the cluster (derived from the generic PLA, Color On Demand or TPU95A)
ColorFabb PLA White
ColorFabb PLA Dark Blue
FormFutura Python Flex Clear
FormFutura EasyFil PLA Magenta

So I should be able to remove the below materials
FormFutura EasyFil PLA Orange
ColorFabb PLA/PHA Magenta
ColorFabb ColorFabb RAL5000

It's not possible, see the attached screenshots.
[image]
[image]
[image]

I hope the whole PLA type materials shouldn't be removed from the cluster in order to allow the materials removals. Doesn't make sense.

Please reopen the issue.

mahtDFR, 2020-05-08:
Hello @V3DPrinting, I answered on the forum already but I'll put it here too: Would it be helpful it we discuss this on a videocall so I can clearly see where the issues are from your side? If we can get some clear reproduction steps, then I think we will be able to debug the issue more efficiently.

GregValiant, 2024-11-22:
I'm cleaning house.
Is this still an issue in current Cura versions (5.8.0 and up)?  Can this be closed?

V3DPrinting, 2024-11-23:
Hi Greg

Thanks for touching base.

The issue seems not to be anymore present in 5.8.

So you can close it

Thanks

Best
David Vandeweghe

Founder – CEO 

***@***.*** ***@***.***>
Mobile : +33 6 74 64 40 93

￼

                                                                                                                        
30 Avenue du Maréchal Gallieni, 78700 Conflans Sainte Honorine

www.v3d-printing.com <http://www.v3d-printing.com/>
This email and any attached files ("Message") may contain confidential and/or privileged information. It is intended solely for the addressee(s).

If you receive this Message in error, inform the sender by reply email, delete the Message and destroy any printed copy.

Any unauthorized use, distribution, or copying of this Message or any part thereof is prohibited. Emails are susceptible to alteration.

Neither V3D Printing nor any of its affiliates shall be liable for the Message if altered or falsified nor shall they be liable for any damage caused by any virus that might be transmitted with this Message.

GregValiant, 2024-11-24:
Another one bites the dust.

closed 2024-11-24 by GregValiant