Ultimaker/Cura #6734: Feature Request: Option to print skin before walls

mazamorac, 2019-11-30:
**Is your feature request related to a problem? Please describe.**
Bridging without supports or sacrificial layers is well handled by Cura if it's on a rectangular overhang. Any other kind of filling an overhang relies on having the first few layers of skin sagging or bunching (usually starting the print in mid-air), and just filling it in by brute force. For an example please see [issue 3513](https://github.com/Ultimaker/Cura/issues/3513).

**Describe the solution you'd like**
A relatively simple solution is to afford the option to print skin before walls. This usually will start printing from the edge, and the wall will adhere to that.

For example, printing the famous articulated octopus in a simulated vase mode, by using 1 layer skins and walls, plus concentric skin, now results in the following trajectory:
[image]

This will result in dropouts and print failure. Printing the skin before the walls will fix this. This scenario is the single most common overhang scenario that I find the most when printing while avoiding supports.

**Describe alternatives you've considered**
The simplest alternative is the use of supports. It's obvious that there are many valid use cases where supports should be avoided. There is also the workaround described in issue 3513, which places the onus of fixing the issue on the designer, and implies creating an FDM-printing-specific model. If we want wide adoption of 3d printing, technology-agnostic modeling should be supported.

**Affected users and/or printers**
All users, all printers.

Ellecross, 2019-12-02:
(for UM cura devs - i'll put this up in the board for discussion)

Ellecross, 2019-12-03:
Found a similar request before: https://github.com/Ultimaker/Cura/issues/2988

Ellecross, 2019-12-04:
Good news! We discussed this today in the team and we appealed to our overlords! We're going to put this in our backlog (which means that it will be further discussed on how we can implement and then we'll put it in our sprint). yeaaaaaaaaahhhhhh <3

Felixoid, 2020-09-22:
Greetings. Is there any way to affect the issue urgency? To me it looks like a possible sollution for the hanging layer's walls as well (see a photo)
[image]

Ghostkeeper, 2020-09-23:
Bringing up arguments to support your cause works.

I don't see how printing the skin before the walls would improve that print though! The walls hardly touch the skin at all for you, it seems. Especially not the outer wall for which that hanging issue occurs. Consider increasing the skin overlap?

Felixoid, 2020-10-05:
Hmm, maybe it's confusing, there are the model on the left and the support on the right part of photo. So, the walls are straight, instead of being arched, on many places.

Not sure, though, if the `skin first` would help. I'm haning around currently to find a proper sollution for this particular issue

Smeik1111, 2020-10-20:
I want this feature too. I have this model which I would like to print [here](https://imgur.com/YGbUWqq), which is an ortholinear keyboard. It has many features on the inside of the model, which either dont stick or get loose with travel-moves as the nozzle scrapes against the outer or inner shell. [Picture](https://imgur.com/Az3JCqr). This can be solved if the skin is printed first (second layer has none of these issues) or the first layer is squished into the bed. I hope this can be added reasonably soon because many models with many features on the inside of the first layer could really profit from that.

fvrmr, 2021-08-18:
The previous ticket mentioned above was removed from our backlog, but there is a new one.
Devs see CURA-7028

nimaet, 2023-07-13:
I've been experimenting with adding fillets to my 3D models to make them printable without the need for support. One thing I've found helpful is leveraging the skin feature in Cura.

In Cura, the default behavior is to print the walls before the skin (top/bottom skin or infill). However, if you're dealing with overhangs and want to ensure that the outer wall is not floating in the air, printing the skin before the wall can be beneficial.

To illustrate my point, I have attached a picture showing an example of a filleted model with an overhang when it is better to print the skin before the wall.
[image]
[image]

sityware, 2023-11-14:
Can I just add my request for this.
I would only need Skin before Walls on the initial layer.

I struggle to tune my printer to remove stringing between holes on the initial layer causing scars on the final print. Like below.

[image]