Ultimaker/Cura #6735: [4.4.0] crash on startup on ubuntu 18.04

n00b3rpwn4g3, 2019-11-30:
**Application version**
4.4.0

**Platform**
Linux-5.0.0-36-generic-x86_64-with-debian-buster-sid

**Printer**
no printer was attached but I had a custom Ender 3 profile loaded from using a previous cura version that I installed from the apt packages

**Reproduction steps**
I downloaded the most recent appimage from Cura's website for Linux distributions, set it to executable, and tried to run it in my downloads folder. It starts up but immediately gives me a crash report.

**Actual results**
It gives me a few separate windows of the same crash report, with more details below.

**Expected results**
It should start up normally I guess.

**Additional information**
This is an image of the crash report window, and I'll post the full error traceback and log parts below:
[image]
error traceback:
```
Traceback (most recent call last):
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/Machines/Models/IntentCategoryModel.py", line 98, in _update
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/Settings/IntentManager.py", line 119, in currentAvailableIntentCategories
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/Settings/IntentManager.py", line 61, in intentCategories
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/Settings/IntentManager.py", line 42, in intentMetadatas
KeyError: 'empty'
```
logs:
```
Thread 0x00007efd99ffb700 (most recent call first):
  File "/tmp/.mount_UltimaTNWt3F/usr/bin/plugins/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efd9a7fc700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1153 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efd9affd700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 549 in wait
  File "/tmp/.mount_UltimaTNWt3F/usr/bin/plugins/plugins/UM3NetworkPrinting/src/Network/ZeroConfClient.py", line 74 in _handleOnServiceChangedRequests
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efd9b7fe700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 297 in wait
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1569 in wait
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 1017 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efd9bfff700 (most recent call first):
  File "/build/env/4.4/inst/lib/python3.5/site-packages/zeroconf.py", line 946 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap

Thread 0x00007efdb0b18700 (most recent call first):
  File "/tmp/.mount_UltimaTNWt3F/usr/bin/plugins/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 86 in _updateThread
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 862 in run
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 914 in _bootstrap_inner
  File "/build/env/4.4/inst/lib/python3.5/threading.py", line 882 in _bootstrap
[... 52 lines cut ...]
  File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 428 in event
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1003 in event
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CrashHandler.py", line 391 in _show
  File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/Event.py", line 189 in call
  File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 428 in event
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1003 in event
  File "/build/4.4/build/inst/lib/python3.5/site-packages/UM/Qt/QtApplication.py", line 341 in exec_
  File "/build/4.4/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 804 in run
  File "/build/4.4/build/inst/bin/cura", line 180 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.4/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
  File "/build/env/4.4/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 957 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 968 in _find_and_load
```
sorry if it's a really long post I just wanted to include as much relevant info.

smartavionics, 2019-11-30:
Hi, I think this is a known bug which I believe I have fixed in my builds so you may wish to try one of those. They can be installed alongside the UM builds. You can find them at https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0. Please read the README.md file in there.

n00b3rpwn4g3, 2019-11-30:
Thank you, I downloaded the 2019-11-28 appimage version and it works without errors.

n00b3rpwn4g3, 2019-11-30:
So should I close the issue now, or does it still kinda apply to the official version so this thread stays open until that's fixed?

smartavionics, 2019-11-30:
Yes, keep this issue open until the UM people confirm that they have fixed the bug in their sources. Cheers!

nallath, 2019-12-02:
We've also fixed it in master.

closed 2019-12-02 by nallath