Ultimaker/Cura #6736: [4.4.0] Won't start on Mac OS Catalina due to "cannot check it for malicious software"

mjparme, 2019-11-30:
**Application version**

4.4.0

**Platform**

Mac OS 10.15.1

**Printer**

Maker Select

**Reproduction steps**

Double-click on the Cura Application

**Actual results**
Get the following error dialog:

[image]

**Expected results**

Cura should start up.

**Additional information**

closed 2019-11-30 by mjparme

smartavionics, 2019-11-30:
Looks like what is discussed in https://www.winytips.com/installer-cant-be-opened-because-apple-cannot-check-it-for-malicious-software/

mjparme, 2019-11-30:
> Looks like what is discussed in https://www.winytips.com/installer-cant-be-opened-because-apple-cannot-check-it-for-malicious-software/

Yeah, I discovered I could convince it to open on the privacy & security pref pane.

SamCB, 2020-01-20:
Can also open it by right clicking->open->open. That said, Apple is definitely making life difficult for Open Source developers right now.

I'm not sure that this issues should be closed. For any mac user, this is now the first thing they see when they try opening the app and for a general consumer who doesn't know to look around on github for workarounds, they're going to have a pretty rough time. What's the process for an open source app like Cura to receive notarization?

wifiprintguy, 2020-06-25:
If you can leverage the info documented here (https://developer.apple.com/documentation/xcode/notarizing_macos_software_before_distribution/customizing_the_notarization_workflow) then you can automate the notarization process. That is what we do in the istopwg/ippeveselfcert project.