Ultimaker/Cura #6737: [4.4.0] Crash on startup

karl-nilsson, 2019-11-30:
**Application version**
4.4.0

**Platform**
Linux-5.3.14_1-x86_64-with-glibc2.3.4

**Printer**
N/A

**Reproduction steps**
1. build cura (with xbps)
2. run cura

**Actual results**
Crash dialog appears

**Expected results**
Cura window appears

**Additional information**
It looks like the CameraTool plugin fails to load, because of an incompatible SDK version. This is probably because Uranium and CuraEngine v4.4.0 have been tagged, but not released.
I deleted all my cura config files to make sure this was not the cause.

Relevant log snippets:

`Traceback (most recent call last):
  File "/usr/bin/cura", line 180, in <module>
    app.run()
  File "/usr/lib/python3.6/site-packages/cura/CuraApplication.py", line 786, in run
    self.runWithGUI()
  File "/usr/lib/python3.6/site-packages/cura/CuraApplication.py", line 852, in runWithGUI
    camera_tool.setOrigin(Vector(0, 100, 0))
AttributeError: 'NoneType' object has no attribute 'setOrigin'`

`[MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [LocalContainerProvider] with supported sdk versions [['6.0.0']] is incompatible with the current sdk version [7.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [FileLogger] with supported sdk versions [['6.0.0']] is incompatible with the current sdk version [7.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [LocalFileOutputDevice] with supported sdk versions [['6.0.0']] is incompatible with the current sdk version [7.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [SelectionTool] with supported sdk versions [['6.0.0']] is incompatible with the current sdk version [7.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [RotateTool] with supported sdk versions [['6.0.0']] is incompatible with the current sdk version [7.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [MirrorTool] with supported sdk versions [['6.0.0']] is incompatible with the current sdk version [7.0.0].
[MainThread] UM.PluginRegistry.loadPlugin [374]: Plugin [CameraTool] with supported sdk versions [['6.0.0']] is incompatible with the current sdk version [7.0.0].`

[gist of full error logs](https://gist.github.com/karl-nilsson/fd8a412d4878d84e350fa9ff56fbcf86
)

nallath, 2019-12-02:
I don't quite know what you did, but your setup does not seem correct. The release as provided on Github already includes Uranium and CuraEngine. As such, we don't make seperate releases for them.

If you want to run from source, you will need to check out the repositories based on the tags.

karl-nilsson, 2019-12-02:
> I don't quite know what you did, but your setup does not seem correct.

I'm the package maintainer for a linux distro, I'm building and packaging cura (and dependencies) from source using the same setup I've used for 2 years. I use the most recent tagged release for each component. I'm sorry I didn't make that clear earlier.

> As such, we don't make seperate releases for them.

In the past, you have released new versions of each component on GitHub:
https://github.com/Ultimaker/Uranium/releases
https://github.com/Ultimaker/CuraEngine/releases

> If you want to run from source, you will need to check out the repositories based on the tags.

Neither Uranium nor CuraEngine have a tag (or release) for v4.4.0, but they do have a corresponding branch. My guess is tagging and releasing would fix the problem.
This is the specific commit that would likely fix the problem. It's present in the 4.4 branch, but not in 4.3.
https://github.com/Ultimaker/Uranium/commit/dd4c574fbbdd31400d8182585a512e4f6f2e3b7c

Ghostkeeper, 2019-12-05:
> Neither Uranium nor CuraEngine have a tag (or release) for v4.4.0, but they do have a corresponding branch. My guess is tagging and releasing would fix the problem.

Sorry about that. This was a problem with miscommunication and holidays. I've tagged them now. There have been no new commits on those branches since release.

Ghostkeeper, 2019-12-05:
The error messages you're seeing seem to indicate that you're using an outdated Uranium version, probably from release 4.3.

karl-nilsson, 2019-12-06:
@Ghostkeeper Thanks! That fixed the problem.
In the future, should I open a bug report for something this small, or just ping someone on IRC?

closed 2019-12-06 by karl-nilsson

Ghostkeeper, 2019-12-06:
Bug reports are more reliable. Pinging someone on IRC can also work but generally I'll see it a few hours later then after you're long gone.

karl-nilsson, 2019-12-06:
ack